mushorg/conpot #337: TypeError: clone() takes at most 2 arguments (4 given)\n

zhiyinyouzhao, 2017-07-27:
Hi，
When I run conpot --template default，I found the error inforamtion in console line.
console output:
Version 0.5.1
  MushMush Foundation

2017-07-27 14:13:36,187 Starting Conpot using template: /usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default
2017-07-27 14:13:36,187 Starting Conpot using configuration found in: /usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/conpot.cfg
WARNING:conpot.utils.ext_ip:Could not fetch public ip from http://whatismyip.akamai.com/
2017-07-27 14:13:42,254 Could not fetch public ip from http://whatismyip.akamai.com/
WARNING:conpot.utils.ext_ip:Could not fetch public ip from http://wgetip.com/
2017-07-27 14:13:48,932 Could not fetch public ip from http://wgetip.com/
WARNING:conpot.utils.ext_ip:Could not fetch public ip: None
2017-07-27 14:13:48,933 Could not fetch public ip: None
2017-07-27 14:13:48,942 Conpot modbus initialized
2017-07-27 14:13:48,942 Found and enabled ('modbus', <class 'conpot.protocols.modbus.modbus_server.ModbusServer'>) protocol.
2017-07-27 14:13:48,951 Conpot S7Comm initialized
2017-07-27 14:13:48,952 Found and enabled ('s7comm', <class 'conpot.protocols.s7comm.s7_server.S7Server'>) protocol.
2017-07-27 14:13:48,957 Found and enabled ('http', <class 'conpot.protocols.http.web_server.HTTPServer'>) protocol.
2017-07-27 14:13:48,962 Found and enabled ('snmp', <class 'conpot.protocols.snmp.snmp_server.SNMPServer'>) protocol.
2017-07-27 14:13:48,970 Conpot Bacnet initialized using the /usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/bacnet/bacnet.xml template.
2017-07-27 14:13:48,971 Found and enabled ('bacnet', <class 'conpot.protocols.bacnet.bacnet_server.BacnetServer'>) protocol.
2017-07-27 14:13:48,979 IPMI BMC initialized.
2017-07-27 14:13:48,979 Conpot IPMI initialized using /usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/ipmi/ipmi.xml template
2017-07-27 14:13:48,980 Found and enabled ('ipmi', <class 'conpot.protocols.ipmi.ipmi_server.IpmiServer'>) protocol.
2017-07-27 14:13:48,981 No proxy template found. Service will remain unconfigured/stopped.
2017-07-27 14:13:48,982 Modbus server started on: ('0.0.0.0', 502)
2017-07-27 14:13:48,982 S7Comm server started on: ('0.0.0.0', 102)
2017-07-27 14:13:48,983 HTTP server started on: ('0.0.0.0', 80)
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/gevent-1.2.2-py2.7-linux-x86_64.egg/gevent/greenlet.py", line 536, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/protocols/snmp/snmp_server.py", line 180, in start
    self.cmd_responder = CommandResponder(host, port, self.compiled_mibs)
  File "/usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/protocols/snmp/command_responder.py", line 55, in __init__
    self.snmpEngine = engine.SnmpEngine()
  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/entity/engine.py", line 90, in __init__
    '__SNMP-FRAMEWORK-MIB', 'snmpEngineMaxMessageSize')
  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 400, in importSymbols
    self.loadModules(modName, **userCtx)
  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 362, in loadModules
    self.loadModule(modName, **userCtx)
  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 330, in loadModule
    'MIB module \"%s\" load error: %s' % (modPath, traceback.format_exception(*sys.exc_info()))
MibLoadError: MIB module "pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.pyc" load error: ['Traceback (most recent call last):\n', '  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 325, in loadModule\n    exec (modData, g)\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.py", line 24, in <module>\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/SNMP-FRAMEWORK-MIB.py", line 71, in clone\n', 'TypeError: clone() takes at most 2 arguments (4 given)\n']
Thu Jul 27 14:13:49 2017 <Greenlet at 0x4e74eb0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x4ed2a90>>('0.0.0.0', 161)> failed with MibLoadError

2017-07-27 14:13:49,042 Bacnet server started on: ('0.0.0.0', 47808)
2017-07-27 14:13:49,043 IPMI server started on: ('0.0.0.0', 623)
ERROR:root:Stopping because <Greenlet at 0x4e74eb0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x4ed2a90>>('0.0.0.0', 161)> died: MIB module "pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.pyc" load error: ['Traceback (most recent call last):\n', '  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 325, in loadModule\n    exec (modData, g)\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.py", line 24, in <module>\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/SNMP-FRAMEWORK-MIB.py", line 71, in clone\n', 'TypeError: clone() takes at most 2 arguments (4 given)\n']
2017-07-27 14:13:49,044 Stopping because <Greenlet at 0x4e74eb0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x4ed2a90>>('0.0.0.0', 161)> died: MIB module "pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.pyc" load error: ['Traceback (most recent call last):\n', '  File "/usr/lib/python2.7/site-packages/pysnmp-4.3.8-py2.7.egg/pysnmp/smi/builder.py", line 325, in loadModule\n    exec (modData, g)\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.py", line 24, in <module>\n', '  File "build/bdist.linux-x86_64/egg/pysnmp/smi/mibs/SNMP-FRAMEWORK-MIB.py", line 71, in clone\n', 'TypeError: clone() takes at most 2 arguments (4 given)\n']

It‘s seems about pysnmp wrong.
pip freeze | grep pysnmp
pysnmp==4.3.8
Please help me solove this issues，thanks.

windflyer, 2017-07-28:
It's odd. I cannot reproduce...

creolis, 2017-07-28:
@zhiyinyouzhao please also provide us with the operating system / derivate / version you run in order to give us a chance to run our tests with your environment :)

zhiyinyouzhao, 2017-08-01:
@windflyer 
pip freeze | grep pysmi
pysmi==0.1.3

zhiyinyouzhao, 2017-08-01:
@creolis 
cat /etc/redhat-release
CentOS Linux release 7.1.1503 (Core）
cat /proc/version
Linux version 3.10.0-229.el7.x86_64 (builder@kbuilder.dev.centos.org) (gcc version 4.8.2 20140120 (Red Hat 4.8.2-16) (GCC) )

windflyer, 2017-08-01:
@zhiyinyouzhao Thanks. It seems that `__SNMP-FRAMEWORK-MIB.pyc` is missing on your machine.

Could you check the following directories:

1. `/usr/local/lib/python2.7/dist-packages/pysnmp/smi/mibs/instances/`
2. `/usr/lib/python2.7/dist-packages/pysnmp/smi/mibs/instances/`

Please also try to re-install `pysnmp` if possible, using the following command:
    
    sudo pip install -U pysnmp==4.3.8

For my desktop running Ubuntu 14.04, the full path of `__SNMP-FRAMEWORK-MIB.pyc` is  `/usr/local/lib/python2.7/dist-packages/pysnmp/smi/mibs/instances/__SNMP-FRAMEWORK-MIB.pyc`

zhiyinyouzhao, 2017-08-03:
@windflyer 
I cheeck some directories .
/usr/lib/python2.7/site-packages/pysnmp/smi/mibs/instances.
__SNMP-FRAMEWORK-MIB.pyc  was in it.
and reinstall pysnmp==4.3.8,the problem is still unsolved.
I think this may be a problem about one of the installation package.
`pip list about requirements.txt  package were here: `
`gevent (1.2.2)
bacpypes (0.16.1)
beautifulsoup4 (4.5.3)
bottle (0.12.13)
crc16 (0.1.1)
cybox (2.1.0.13)
enum (0.4.6)
gevent (1.2.2)
hpfeeds (1.0)
Jinja2 (2.9.6)
requests (2.11.1)
lxml (3.3.5)
libtaxii (1.1.110)
mixbox (1.0.2)
modbus-tk (0.5.7)
MySQL-python (1.2.5)
pyghmi (1.0.22)
pysmi (0.1.3)
pysnmp (4.3.8)
Sphinx (1.6.3)
stix (1.2.0.2)
stix-validator (2.5.0)
xlrd (1.0.0)

`
What other information do I need to provide?
Thanks for your support.

zhiyinyouzhao, 2017-08-04:
@creolis 
when I run conpot with Ubuntu 15.10 LTS.
It‘s function normal.
So,I thank this bug maybe have something to do with Centos 7.1.
You can fix bug with this way about operating system.
Thank you very much!

windflyer, 2017-08-04:
@zhiyinyouzhao I'm actually trying to reproduce this bug in the CentOS VM I just created. 
I think most of us do conpot development on Debian-based systems. I'm sorry we were not able to cover all Linux distributions at this point.

Vingaard, 2017-08-04:
Hi all, just a suggestion for further troubleshooting, since we are talking about Centos vs. Ubuntu. 
Try to pull a list of the installations files (with version numbers) on both the Ubuntu and the Centos environment.. There 'might' be some differences on the version number in one or more packages , and this could potential be a part of the problem, just my two cents.

windflyer, 2017-08-04:
@Vingaard You're right. `pysnmp` 4.3.8 works perfectly on Ubuntu but breaks on CentOS.

@zhiyinyouzhao On CentOS, please update your local `requirements.txt` and use the version number 4.3.9 for `pysnmp`. The line of pysnmp should be like:

    pysnmp==4.3.9

The issue was fixed for my CentOS 7 VM.

Vingaard, 2017-08-04:
@windflyer many thanks for fast feedback

zhiyinyouzhao, 2017-08-05:
@windflyer @Vingaard 
I try to update pysnmp==4.3.9 in Centos.
So far, the problem has been solved ,it‘s  perfectly.
Thanks a lot.

windflyer, 2017-08-25:
We already have composed a few installation instructions here https://github.com/mushorg/conpot/tree/master/docs/source/installation

I think we are good to close this issue for now...

closed 2017-08-25 by creolis