mushorg/conpot #339: SCAN S7 PLC filtered

Hemza, 2017-08-10:
Hi All;

i've try to scan PLCs, but seems it's located behind a Firewall and the Port 102 is filteres, see the scan results ::
-----------------------------------------------------------------------------------
sudo nmap -Pn -p 102 --script s7-enumerate -sV Target_IP

Starting Nmap 7.40 ( https://nmap.org ) at 2017-08-10 10:24 EDT
Nmap scan report for Target_IP
Host is up.
PORT    STATE    SERVICE  VERSION
102/tcp filtered iso-tsap

Service detection performed. Please report any incorrect results at https://nmap.org/submit/ .
Nmap done: 1 IP address (1 host up) scanned in 2.39 seconds
-------------------------------------------------------------------------------------------

Is there any other method to SCAn PLCs  ?? i've tried the -sI nmap options and -D but the same result is returned;

creolis, 2017-08-10:
Hey Hemza, 
what PLC are you trying to scan? Is it a real PLC or is it a conpot instance?
Who is the owner of the PLC (or the conpot instance) ?

If nmap replies "filtered" on a TCP port, this tells you that while receiving
no SYN-ACK from the remote peer, it also got no ICMP destination port unreachable - so either the remote peer or an active network filtering component between you and the remote peer chose to silently drop your connection attempt(s).

creolis, 2017-08-25:
It seems you the question is answered (or not longer relevant) for you. If you need further assistance, feel free to re-open the issue again.

closed 2017-08-25 by creolis