mushorg/conpot #340: Docker build fails

jamz212, 2017-08-11:
Prebuild docker image works as expected, but not sure how to get it to autorun at boot. Trying to compensate by building the image locally but it fail with error below. Would appreciate any help with this issue.  

FAILED (failures=1, errors=2, skipped=3)
Test failed: <unittest.runner.TextTestResult run=34 errors=2 failures=1>
Coverage.py warning: Trace function changed, measurement is likely wrong: <bound method PyTracer._trace of <PyTracer at 140673785728528: 2251 lines in 61 files>> (trace-changed)
error: Test failed: <unittest.runner.TextTestResult run=34 errors=2 failures=1>
ERROR: Service 'conpot' failed to build: The command '/bin/sh -c coverage run --timid --source=conpot setup.py test' returned a non-zero code: 1

windflyer, 2017-08-25:
Hi @jamz212 to quickly solve this problem, please comment out line 24 

    RUN coverage run --timid --source=conpot setup.py tes

Seems that our test cases would fail in docker environment. Will take a closer look at it.

creolis, 2018-09-10:
fixed in current releases

closed 2018-09-10 by creolis