mushorg/conpot #343: Automatic randomization

ddavidebor, 2017-09-05:
Following #291 , it seems to me conpot needs to be customized manually to fool someone. Is this right ?

If so, i'd strongly suggested randomizing at startup time the configuration. It should catch the flies by default.

glaslos, 2017-09-05:
This has been discussed many times and depending on the wind direction we are leaning either left or right. Pro randomizing is making sure a vanilla Conpot is not easily detected (adversaries and SHODAN), pro current status is we don't mind vanilla Conpot being detected. Con randomizing is the lack of a leaning effect, con current status is the bad reputation for the project.
What is your argument for randomizing the default template?

adepasquale, 2017-09-05:
How about going forward with #291 (stop running if default configuration is detected) :question: and also providing two additional command line switches:
* one for forcing to start anyway, e.g. `-f`;
* one for randomizing default configuration.

Vingaard, 2017-09-05:
Welcome @ddavidebor . just my two cents - I do see your point (and agree!)  but detecting Conpot are still trivial,  when you can e.g throw a function 43 into the Modbus service. 
Personal I would like to see that Conpot did offer the  randomizing option - the it's up to the owner if he/she would use the option. BTW, To make effective camoflage, I would suggest to deply the Conpot in proxy mode, connecting to a real device.

creolis, 2017-09-06:
I'm with @adepasquale - let's go on with #291 and refuse to start with default template (if you dont "force" the default start using an option switch that shows that you know what you're doing).

I understand the concerns of @ddavidebor, but an industrial control system honeypot is not something you buy in a grocery store, plug-in and turn a crank - you need to think what you want to achieve, what values actually make sense for your specific use case so that you're providing a consistent picture. You'll never provide a sufficient HTTP interface "out of the box" without the risk of being detected. You have to consider blocking other ports on your box from being detected (e.g. SSH, etc.). So I'm not sure if we do ourselves a favor in stating that we would provide a hard-to-detect randomized behaviour out of the box.

As @Vingaard stated, I'm okay with _offering_ such a feature to some extent, but only to give you a starting point. I would not consider a "randomized" databus as harder to detect. We're pretty much low interaction after all, so even little effort (like using unsupported function codes) would be sufficient anyway :)

glaslos, 2017-09-10:
Sounds good, let's go with #291 then for now and keep this discussion open for the future.

creolis, 2018-01-14:
#291 is now closed.

closed 2018-08-02 by xandfury