mushorg/conpot #347: Can a real PLC be linked with Conpot instead of Simulating one? I run it on TPOT.

SwapnilDreams100, 2017-11-02:


glaslos, 2017-11-03:
You can use the proxy module to attach a real PLC. https://github.com/mushorg/conpot/tree/master/conpot/templates/proxy

SwapnilDreams100, 2017-11-04:
Thanks Ill try it out. What changes do I have to make? I will be using a Seimens S7 1200.

SwapnilDreams100, 2018-01-31:
Hi, can you please list the changes to be done/ steps to be followed for liking the real plc to conpot. Highly appreciated.

glaslos, 2018-02-04:
Hi @SwapnilDreams100 have a look at the [configuration](https://github.com/mushorg/conpot/blob/master/conpot/templates/proxy/proxy/proxy.xml) in the Proxy sample template. You can see how we map 0.0.0.0:dest_port to proxy_host:proxy_port. If you set the proxy host and port to your S7, you should see the connections in Conpot and in your PLC.

xandfury, 2018-06-12:
Question seems to be answered. @SwapnilDreams100 If you need any further assistance, feel free to re-open the issue again.

closed 2018-06-12 by xandfury