mushorg/conpot #348: EtherNet/IP template

glaslos, 2017-11-18:
Add a template that is using the EtherNet/IP protocol stack.

Sumit-ai-dev, 2026-10-11:
Hi @glaslos,

Was looking through the recent TOML refactor across templates (like `ads` and `opcua`), and noticed `enip` is still only configured inside `default`—so `conpot --template enip` doesn't work yet.

I'm thinking of adding a standalone `templates/enip/` profile (`template.toml` + `enip.toml`) following the same pattern, along with tests in `test_toml_templates.py` and a doc update in `enip.rst`.

A couple of quick thoughts on design:
1. **Vendor in `template.toml`**: `"Rockwell Automation"` or `"Generic"`?
2. **Tags**: Stick to the 3 baseline tags from `default`, or add a few common PLC types like `BOOL`/`DINT`?

Does this approach sound good? Happy to be assigned and put in a PR!

glaslos, 2026-10-11:
Awesome!
1. Either is fine, they can both be finger printed.
2. Feel free to add new types 👍

closed 2026-10-11 by glaslos