mushorg/conpot #351: <Greenlet at 0x7f03ad2714b0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x7f03ad2c45d0>>('0.0.0.0', 161)> failed with WrongValueError

sepantakntu, 2017-12-19:
hi 
how can help me to solve this error?
root@ubuntu:/opt/conpot# conpot --template default

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.5.1
  MushMush Foundation

2017-12-19 06:51:01,971 Starting Conpot using template: /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default
2017-12-19 06:51:01,972 Starting Conpot using configuration found in: /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/conpot.cfg
2017-12-19 06:51:02,146 Starting new HTTP connection (1): whatismyip.akamai.com
2017-12-19 06:51:05,538 Fetched 46.209.11.34 as external ip.
ERROR:conpot.protocols.modbus.modbus_server:'module' object has no attribute 'REPORT_SLAVE_ID'
2017-12-19 06:51:05,540 'module' object has no attribute 'REPORT_SLAVE_ID'
2017-12-19 06:51:05,540 Found and enabled ('modbus', <class conpot.protocols.modbus.modbus_server.ModbusServer at 0x7f03adcc4bb0>) protocol.
2017-12-19 06:51:05,541 Conpot S7Comm initialized
2017-12-19 06:51:05,541 Found and enabled ('s7comm', <class 'conpot.protocols.s7comm.s7_server.S7Server'>) protocol.
2017-12-19 06:51:05,542 Found and enabled ('http', <class 'conpot.protocols.http.web_server.HTTPServer'>) protocol.
2017-12-19 06:51:05,543 Found and enabled ('snmp', <class 'conpot.protocols.snmp.snmp_server.SNMPServer'>) protocol.
2017-12-19 06:51:05,544 Conpot Bacnet initialized using the /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/bacnet/bacnet.xml template.
2017-12-19 06:51:05,544 Found and enabled ('bacnet', <class 'conpot.protocols.bacnet.bacnet_server.BacnetServer'>) protocol.
2017-12-19 06:51:05,545 IPMI BMC initialized.
2017-12-19 06:51:05,545 Conpot IPMI initialized using /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/ipmi/ipmi.xml template
2017-12-19 06:51:05,546 Found and enabled ('ipmi', <class 'conpot.protocols.ipmi.ipmi_server.IpmiServer'>) protocol.
2017-12-19 06:51:05,547 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2017-12-19 06:51:05,547 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2017-12-19 06:51:05,547 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2017-12-19 06:51:05,548 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
2017-12-19 06:51:05,548 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2017-12-19 06:51:05,548 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2017-12-19 06:51:05,548 Found and enabled ('enip', <class 'conpot.protocols.enip.enip_server.EnipServer'>) protocol.
2017-12-19 06:51:05,548 No proxy template found. Service will remain unconfigured/stopped.
2017-12-19 06:51:05,548 Modbus server started on: ('0.0.0.0', 502)
2017-12-19 06:51:05,549 S7Comm server started on: ('0.0.0.0', 102)
2017-12-19 06:51:05,554 HTTP server started on: ('0.0.0.0', 80)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent/greenlet.py", line 536, in run
    result = self._run(*self.args, **self.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/protocols/snmp/snmp_server.py", line 180, in start
    self.cmd_responder = CommandResponder(host, port, self.compiled_mibs)
  File "/usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/protocols/snmp/command_responder.py", line 78, in __init__
    config.addV1System(self.snmpEngine, 'public-read', 'public')
  File "/usr/local/lib/python2.7/dist-packages/pysnmp-4.3.9-py2.7.egg/pysnmp/entity/config.py", line 78, in addV1System
    ((snmpCommunityEntry.name + (8,) + tblIdx, 'destroy'),)
  File "/usr/local/lib/python2.7/dist-packages/pysnmp-4.3.9-py2.7.egg/pysnmp/smi/instrum.py", line 256, in writeVars
    return self.flipFlopFsm(self.fsmWriteVar, varBinds, acInfo)
  File "/usr/local/lib/python2.7/dist-packages/pysnmp-4.3.9-py2.7.egg/pysnmp/smi/instrum.py", line 239, in flipFlopFsm
    raise origExc
WrongValueError: WrongValueError({'msg': WrongValueError(), 'name': (1, 3, 6, 1, 6, 3, 18, 1, 1, 1, 4, 112, 117, 98, 108, 105, 99, 45, 114, 101, 97, 100), 'idx': 0})
Tue Dec 19 06:51:05 2017 <Greenlet at 0x7f03ad2714b0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x7f03ad2c45d0>>('0.0.0.0', 161)> failed with WrongValueError

2017-12-19 06:51:05,896 Bacnet server started on: ('0.0.0.0', 47808)
2017-12-19 06:51:05,896 IPMI server started on: ('0.0.0.0', 623)
2017-12-19 06:51:05,897 handle server PID [ 3188] running on ('0.0.0.0', 44818)
2017-12-19 06:51:05,897 handle server PID [ 3188] responding to external done/disable signal in object 139653767458448
ERROR:root:Stopping because <Greenlet at 0x7f03ad2714b0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x7f03ad2c45d0>>('0.0.0.0', 161)> died: WrongValueError({'msg': WrongValueError(), 'name': (1, 3, 6, 1, 6, 3, 18, 1, 1, 1, 4, 112, 117, 98, 108, 105, 99, 45, 114, 101, 97, 100), 'idx': 0})
2017-12-19 06:51:05,897 Stopping because <Greenlet at 0x7f03ad2714b0: <bound method SNMPServer.start of <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x7f03ad2c45d0>>('0.0.0.0', 161)> died: WrongValueError({'msg': WrongValueError(), 'name': (1, 3, 6, 1, 6, 3, 18, 1, 1, 1, 4, 112, 117, 98, 108, 105, 99, 45, 114, 101, 97, 100), 'idx': 0})

dbellavista, 2018-01-04:
The `pysnmp` version used by conpot is no longer working. Upgrading to `4.4.4` in `requirements.txt` seems to solve. However I didn't perform full tests.

sfigue, 2018-01-09:
Hi,
I tried to use conpot, since tpot 17.10. When i ran the default template with (conpot -t default) sended me the same issue that @sepantakntu.
Then, i followed @dbellavista  recomendation and i tried to upgrade pysnmp with (easy_install --upgrade pysnmp) and had a new issue (Unable to execute 'gcc': No such file or directory). 
Then i applied (apk add --no-cache gcc) and i had a new issue (gcc:stdio.h: No such file or directory).

Any solution?

Thanks.

dbellavista, 2018-01-09:
@sfigue I guess it's related with a missing dependency in `Alpine` not with conpot

sfigue, 2018-01-09:
@dbellavista, You are right. Tpot 17.10 docker images, have been rebuilt as micro containers based on Alpine. But, all begun with the same error that @sepantakntu.

dbellavista, 2018-01-09:
I use a Dockerfile similar to https://github.com/mushorg/conpot/blob/master/Dockerfile:
* Edit in `requirements.txt` the `pysnmp` version to `4.4.4`
* `pip install -r requirements.txt`
* `python setup.py install`

sfigue, 2018-01-11:
@dbellavista thank you for the input. Here, a solution (Tpot community help me):

Conpot is designed to run a certain [configuration set](https://github.com/dtag-dev-sec/tpotce/tree/master/docker/conpot/dist) within T-Pot. Changes to that config will involve changes to:

1. /opt/tpot/tpot.yml (since different ports will be used for container start & adding a volume for the container config where to load your config from)
2. /etc/systemd/system/tpot.service (since iptables exclusions need to be modified according to the ports of the Conpot config changes)

Best starting point will be Conpot's [Dockerfile](https://github.com/dtag-dev-sec/tpotce/tree/master/docker/conpot).

creolis, 2018-01-14:
As a conclusion - this was a tpot related issue and not an issue of conpot vanilla?
Just to be sure that there's nothing to be handled for now :)

dbellavista, 2018-01-15:
@creolis vanilla conpot, some dependencies need to be upgraded

xandfury, 2018-01-20:
Hi,

Please try installing and running Conpot again. I think this issue has been fixed with the lastest few commits.

Vingaard, 2018-01-25:
@xandfury - it may not have been fixed 
I had to set pysnmp>=4.4.4 in requirements.txt before it worked (installed this evening, plain conpot, no TPOT)

xandfury, 2018-02-16:
@Vingaard I suspect you are running multiple version of `pysnmp`. If you already have a version of pysnmp installed, a cached version of the requirement is used. Could you please paste the output of the following:

```
python -c 'import pysnmp; [path] = pysnmp.__path__; print(path)'  ## this should print the path of pysnmp you are using.
```
The output of the above command would be something like:
```
/usr/local/lib/python2.7/dist-packages/pysnmp
```
Then `ls` a directory above. Like for the above output, do 
```
ls  /usr/local/lib/python2.7/dist-packages/ | grep pysnmp
```
Notice no `/pysnmp` at the end of `/usr/local/lib/python2.7/dist-packages/`. 
This should give you versions of snmp installed. Such as:
```
pysnmp-4.3.9.dist-info
pysnmp-4.2.5.dist-info
```

Vingaard, 2018-02-22:
@xandfury - thanks for your message - however i am not the OP, that was @sepantakntu 
I just clarified that the problem was present and provided a work-around for future viewers