mushorg/conpot #352: Add FTP/TFTP support

xandfury, 2018-01-01:
Many of the modern meters/plc/rtu provide support for FTP. 
Some links:
Sofrel Lacroix S500 Range RTU's
- https://www.shodan.io/search?query=direct+comments+to+integrated+systems+sofrel

Schneider Electric's PowerLogic™ PM5500 series
- http://docs-europe.electrocomponents.com/webdocs/1568/0900766b815685bf.pdf
Please check page 74

ABB AC500 series:
- https://www.shodan.io/search?query=ac500+ftp+server+ready
- https://search-ext.abb.com/library/Download.aspx?DocumentID=3ADR020077C0204&LanguageCode=en&DocumentPartId=&Action=Launch

xandfury, 2018-01-01:
Python's `ftplib` may be used to facilitate emulation.

creolis, 2018-01-10:
Sounds great - we should try to allow a high level of customization there.
Next to active and passive mode, we should allow to specify the commands supported (see: http://www.nsftools.com/tips/RawFTP.htm) as well as how each command individually responds / works, in order to flexibly mimick each desired system.

I did not look at ftplib yet and therefore did not check if this would satisfy this :-)

SwapnilDreams100, 2018-02-21:
Hi I would like to begin on this project, can anyone give me more links to study the details of what needs to be implemented? I know the basics of protocols like telnet and ftp

creolis, 2018-08-02:
Implemented by @xandfury in the upcoming release

closed 2018-08-02 by creolis