mushorg/conpot #353: Conpot as a Modbus TCP-RTU bridge

xandfury, 2018-01-20:
Current implementation of Modbus uses a predefined template constisting of coils/registers - related values. It might be a good idea to be able to add actual modbus rtu slaves to conpot. 

For each request from a modbus client, Conpot would query the actual rtu slave device(that is connected via serial) and fetch the appropriate response.

***What needs to be done***:
This would involve Conpot to act as RTU master as well as the TCP server. Conversion of TCP to RTU involves dropping the MBAP header and adding CRC. (opposite for RTU to TCP)

***How can it be done***: 
`modbus_tk` supports both Modbus RTU and Modbus TCP. With the additional dependency of `pyserial`, this seems achievable. For integration to exisiting protocol stack, I am not sure about the logical way to approach this. But I think instead of `_configure_slaves(template)` and `_get_mode_and_delay(template)` approriate code need to be added to interface with the existing Modbus TCP server.

xandfury, 2018-01-23:
Something like this in the template:

```
    <slaves>
        <slave id="0">
               ~~~ From template ~~~
        </slave>
        <slave id="2">
               ~~~ Actual Modbus RTU slave connected over serial ~~~~
         </slave>
        <slave id="3">
               ~~~ From template ~~~
        </slave>
        .
        .
        .
    </slaves>
```

glaslos, 2018-01-26:
First goal should be talking to a Modbus slave using Modbus over TCP.

glaslos, 2026-09-14:
## Verification (2026-09-14)

This was still open and **not implemented**. The existing Modbus `mode=serial` only changes addressing semantics over TCP (broadcast UID 0, etc.); it does not talk to real RTU/TCP backends. Prior WIP in #356 (serial-to-ethernet converter) was closed without merging.

## Fix prepared locally (not yet committed)

Per-slave `<remote type="tcp|rtu">` bridging as sketched in this issue / the follow-up comment:

- Mix template-backed `<blocks>` slaves with remote slaves in one template
- TCP first (as discussed), plus RTU via `modbus_tk` + `pyserial` (already pulled in by `modbus-tk`)
- Shared master per TCP endpoint / serial port; PDU-level forward (drop MBAP / add CRC for RTU)
- XSD + docs (`customization.rst`) updated; tests in `test_modbus_remote.py`

Leaving this issue open until the change is reviewed/merged.

glaslos, 2026-09-14:
## Update: internal slaves (not external bridge)

Redirected from proxying to real TCP/RTU devices. Conpot now treats issue #353 as **template-backed internal multi-slave** support:

- Removed any external `<remote>` / backend-master approach
- In `mode=tcp`, every configured `<slave id="…">` is addressable by that unit id (previously only 0/255 were served; other IDs always returned slave-device failure)
- `mode=serial` broadcast/1–247 behaviour unchanged
- Docs and tests cover independent internal unit ids over Modbus/TCP

Still uncommitted locally until review.

glaslos, 2026-09-14:
f2c51686bb1ba774142c2aed75d9ab79dd61fda5 introduces template-backed slave IDs. I decided not to implement serial bridging to external slaves due to the security implications.

closed 2026-09-14 by glaslos