mushorg/conpot #354: SNMP not Responding

dirtdiver512, 2018-02-02:
I have a basic install.  port 80 responses but can not get SNMP to do anything.  Nothing in the logs either.

sudo conpot -f -t default -c /home/conpot/conpot/testing.cfg  -l /var/log/conpot/conpot2.log

2018-02-01 15:53:15,064 --force option specified. Using testing configuration:
/usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/testing.cfg
2018-02-01 15:53:15,067 Starting Conpot using template: /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default
2018-02-01 15:53:15,067 Starting Conpot using configuration found in: /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/testing.cfg
2018-02-01 15:53:15,204 Fetched 209.165.134.1 as external ip.
ERROR:conpot.protocols.modbus.modbus_server:'module' object has no attribute 'REPORT_SLAVE_ID'
2018-02-01 15:53:15,210 'module' object has no attribute 'REPORT_SLAVE_ID'
2018-02-01 15:53:15,212 Found and enabled ('modbus', <class conpot.protocols.modbus.modbus_server.ModbusServer at 0x7fb2bd3f4940>) protocol.
2018-02-01 15:53:15,218 Conpot S7Comm initialized
2018-02-01 15:53:15,219 Found and enabled ('s7comm', <class 'conpot.protocols.s7comm.s7_server.S7Server'>) protocol.
2018-02-01 15:53:15,222 Found and enabled ('http', <class 'conpot.protocols.http.web_server.HTTPServer'>) protocol.
2018-02-01 15:53:15,225 Found and enabled ('snmp', <class 'conpot.protocols.snmp.snmp_server.SNMPServer'>) protocol.
2018-02-01 15:53:15,229 Conpot Bacnet initialized using the /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/bacnet/bacnet.xml template.
2018-02-01 15:53:15,229 Found and enabled ('bacnet', <class 'conpot.protocols.bacnet.bacnet_server.BacnetServer'>) protocol.
2018-02-01 15:53:15,233 IPMI BMC initialized.
2018-02-01 15:53:15,233 Conpot IPMI initialized using /usr/local/lib/python2.7/dist-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/ipmi/ipmi.xml template
2018-02-01 15:53:15,234 Found and enabled ('ipmi', <class 'conpot.protocols.ipmi.ipmi_server.IpmiServer'>) protocol.
2018-02-01 15:53:15,238 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-02-01 15:53:15,239 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2018-02-01 15:53:15,239 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-02-01 15:53:15,241 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
2018-02-01 15:53:15,242 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2018-02-01 15:53:15,242 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-02-01 15:53:15,243 Found and enabled ('enip', <class 'conpot.protocols.enip.enip_server.EnipServer'>) protocol.
2018-02-01 15:53:15,244 No proxy template found. Service will remain unconfigured/stopped.
2018-02-01 15:53:15,245 Modbus server started on: ('0.0.0.0', 502)
2018-02-01 15:53:15,246 S7Comm server started on: ('0.0.0.0', 102)
2018-02-01 15:53:15,247 HTTP server started on: ('0.0.0.0', 80)
2018-02-01 15:53:15,754 SNMP server started on: ('0.0.0.0', 161)
2018-02-01 15:53:15,757 Bacnet server started on: ('0.0.0.0', 47808)
2018-02-01 15:53:15,758 IPMI server started on: ('0.0.0.0', 623)
2018-02-01 15:53:15,759 handle server PID [15723] running on ('0.0.0.0', 44818)
2018-02-01 15:53:15,760 handle server PID [15723] responding to external done/disable signal in object 140405627974320
2018-02-01 15:53:20,204 Privileges dropped, running as "nobody:nogroup"

glaslos, 2018-02-04:
Can you check `netstat -tlpn` if Conpot is listening on port 161 for SNMP?
Probably unrelated but you want to check the modbustk library installed:
```
ERROR:conpot.protocols.modbus.modbus_server:'module' object has no attribute 'REPORT_SLAVE_ID'
2018-02-01 15:53:15,210 'module' object has no attribute 'REPORT_SLAVE_ID'
```

dirtdiver512, 2018-02-06:
yeah port udp 161 is listening.....i can see the packets hitting it via tcpdump but there is nothing in conpot logs

xandfury, 2018-02-16:
Try attempting a simple `snmpwalk` on conpot. Could you also provide some us some info to reproduce the issue? (Pysnmp version, Conpot version, OS version etc.)

xandfury, 2018-08-12:
Closing this issue assuming that the error no longer persists with the latest release. Please feel free to reopen this issue in case you think otherwise.

closed 2018-08-12 by xandfury