mushorg/conpot #355: Unable to collect logs

surajpkhetani, 2018-02-09:
I am unable to collect logs in these files: /var/log/conpot/conpot.log and /var/log/conpot.json.
I have manually created /var/log/conpot/conpot.log with world writable permissions.

This is what the Dockerfile looks like in which gives the --logfile option :
# Create directories
RUN mkdir -p /var/log/conpot/

VOLUME /var/log/conpot/

EXPOSE 80 102 161/udp 502

CMD ["/usr/local/bin/conpot", "--force", "--template", "default", "--logfile", "/var/log/conpot/conpot.log"]

and this how the enabled log file testing.cfg looks like:
[json]
enabled = True
filename = /var/log/conpot.json

I would like to capture these logs generated on the conpot interface:
2018-02-08 17:11:12,306 New snmp session from 10.1.1.129 (da6a2e0b-5d16-47c8-a04a-8eef6e5bed76)
2018-02-08 17:11:12,309 SNMPv1 Get request from ('10.1.1.129', 53024): 1.3.6.1.2.1.1.5.0 
2018-02-08 17:11:12,310 SNMPv1 Get response to ('10.1.1.129', 53024): 1.3.6.1.2.1.1.5.0 CP 443-1 EX40
2018-02-08 17:11:12,497 SNMPv1 Get request from ('10.1.1.129', 53024): 1.3.6.1.2.1.1.1.0 
2018-02-08 17:11:12,497 SNMPv1 Get response to ('10.1.1.129', 53024): 1.3.6.1.2.1.1.1.0 Siemens, SIMATIC, S7-200
2018-02-08 17:11:12,623 SNMPv1 Get request from ('10.1.1.129', 53024): 1.3.6.1.2.1.1.4.0 
2018-02-08 17:11:12,624 SNMPv1 Get response to ('10.1.1.129', 53024): 1.3.6.1.2.1.1.4.0 Siemens AG
2018-02-08 17:11:12,832 SNMPv1 Get request from ('10.1.1.129', 53024): 1.3.6.1.2.1.1.6.0 
2018-02-08 17:11:12,834 SNMPv1 Get response to ('10.1.1.129', 53024): 1.3.6.1.2.1.1.6.0 Venus
2018-02-08 17:11:12,964 SNMPv1 Get request from ('10.1.1.129', 53024): 1.3.6.1.2.1.1.3.0

xandfury, 2018-02-09:
Hey @surajpkhetani 

Only directories mentioned in the `Dockerfile` with `VOLUME` option is treated as persistent storage. As indicated by the `Dockerfile`, files stored in location `/var/log/conpot/` would be treated as part of the volume. 
Persistent volumes are located in `/var/lib/docker/volumes/..` directory of your ***host machine*** (might require you to have `sudo` privileges.) 

Note that `conpot.json` file is located in your `/var/log/` directory. 

This leaves you with 2 choices:

- Either change the location of your json filename from `/var/log/conpot.json` to `/var/log/conpot/conpot.json` and find your file somewhere in `/var/lib/docker/volumes/..` directory. Or, 
- Use docker's `-v` option to mount a local directory as volume on the container. To do this you wouldn't need to change the config. Just create a new volume with `docker volume create --name conpot_logs` and start Conpot with `-v conpot_logs:/var/log/` args appended to your `docker run` command. This would mount container's `/var/log` directory to a newly created volume called `conpot_logs`.
You would find your file at : `/var/lib/docker/volumes/conpot_logs/_data/conpot.json` of your ***host machine***

surajpkhetani, 2018-02-09:
Thanks @xandfury I can now collect the logs using both the methods you mentioned.

closed 2018-02-09 by surajpkhetani

raphkasa, 2019-08-15:
I tried with above choices but no luck. might be i'm doing it in wrong way. Could you please help me:
Attached the screenshot of error
[image]