mushorg/conpot #357: Good exception, Bad exception

xandfury, 2018-02-16:
Bad exception
```
try:
    ...
except FooException as e:                    # Nope
    logger.error("Uh oh! Error: %s", e) 
```
Good exception
```
try:
    ...
except FooException:
    logger.exception("Uh oh!") 
```
Formatting an exception throws away the traceback, which is usually what we want for debugging. logger.exception is a helper that calls logger.error but also logs the traceback.
Unless we using the exception object for handling `socket.errors` or similar, we don't need the object.

gauravzqa, 2018-12-26:
Hi @xandfury i would like to work on this issue and can you help me how to get started, i am new to Open Source.

xandfury, 2018-12-27:
@gauravzqa You can start here:
https://github.com/mushorg/conpot/blob/eeabbbf1067f6c5e0b4a8dfc9314215731b447f8/bin/conpot#L72

Change this to `logger.exception`. Make sure that all tests pass before opening a pull request :slightly_smiling_face: 

If you are a student, you can join [image]. You can get additional feedback from past GSoC mentors at #conpot.

gauravzqa, 2018-12-27:
thanks @xandfury .

gauravzqa, 2018-12-27:
@xandfury what are the tests that you've mentioned above.

gauravzqa, 2018-12-28:
need some help @xandfury 
i am having some problem with tox.
ERROR: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available. and then when i am installing ssl, i am getting many other errors 
error1: ImportError: No module named '_ssl'
error2: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ViQk7c/ssl/.

xandfury, 2018-12-28:
@gauravzqa I don't think this is an issue with conpot. Were you able to install and run it? As far as the errors are concerned, Google is your friend :slightly_smiling_face: 

There are dozens of SE pages that could help you with this.

closed 2019-01-09 by glaslos