mushorg/conpot #358: Port to python3

xandfury, 2018-02-16:
Just putting this out there. No pressure :-)

xandfury, 2018-03-27:
I did some digging and found the latest stable releases for the requirements specified in the master branch. Here are the results:

| Package Name  | py3 support available?  | Latest py3 supported release  |
| ------------ | ------------ | ------------ |
| BACpypes  | Yes  | 0.16.7  |
| beautifulsoup4  | Yes  | 4.6.0  |
|  bottle | Yes  | 0.12.13  |
| cpppo  | Yes  | -  |
| crc16  | -  | py3 successor available  |
| cybox  | Yes  | 2.1.0.17  |
| enum34  | Yes  | 1.1.6  |
| gevent  | Yes  | 1.2.2  |
| hpfeeds  |   | Alternate package called hpfeed3 available  |
| Jinja2  | Yes  | 2.10  |
| mixbox  | Yes  | 1.0.3  |
| modbus_tk  | - | port to python3 in progress  |
| libtaxii  | Yes  | 1.0.3  |
| MySQL-python  | No |  -  |
| natsort  | Yes  | 5.2.0  |
| pyghmi  | Yes  | -  |
| pysmi  | Yes  | 0.2.2  |
| pysnmp  | Yes  | 4.4.4  |
| requests  | Yes  | 2.18.4  |
| scapy  | - | py3 successor available  |
| Sphinx  | Yes  | 1.7.2  |
| stix  | Yes  | 1.2.0.5  |
| stix-validator  | Yes  | -  |
| xlrd  |  Yes | 1.1.0  ||

Hence the only roadblock I find is (***drum-rolls***) MySQL-python. I think this can be overlooked/substituted with some other package. With python2.7 depreciating in Jan 2020, we should also start the python3 port.

creolis, 2018-03-28:
oooof course the package I introduced to conpot is the co... er .. roadblock.
Well .. starting a Python3 branch won't hurt, that's for sure. Doing a last-minute transition will help nobody, so this would give us a head start for refactoring and testing ..

glaslos, 2018-04-01:
MySQL should be no problem. I also wouldn't mind to drop MySQL support or replace it with PostgreSQL. I'm actually more concerned about modbus_tk. Has anyone looked into how far they are into their py3 implementation?

xandfury, 2018-04-01:
@glaslos Check this: https://github.com/ljean/modbus-tk/issues/49

xandfury, 2018-04-05:
In any case, we should atleast aim to double our coverage before attempting anything. Good coverage would save us a lot of pain fixing functionality bugs. 

Note to self: A quick look tells me, we don't have `test_guardian_ast_server`; Might be a good start for this. :-)

creolis, 2018-08-02:
Implemented by @xandfury in the upcoming release

closed 2018-08-02 by creolis