mushorg/conpot #361: Please monkey patch earlier ...

t3chn0m4g3, 2018-03-09:
I recently noticed the following warning when starting conpot (cloned from latest master):

```
/usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/EGG-INFO/scripts/conpot:21: RuntimeWarning: Monkey-patching ssl after ssl has already been imported may lead to errors, including RecursionError on Python 3.6. Please monkey-patch earlier.
gevent.monkey.patch_all()
```

Seems the wrapper is the reason as running `/usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/bin/conpot` works just fine.

I do not know how the wrapper will be created, however, modifying it accordingly ...
```
#!/usr/bin/python
# EASY-INSTALL-SCRIPT: 'Conpot==0.5.1','conpot'
__import__('gevent.monkey').monkey.patch_all()
__requires__ = 'Conpot==0.5.1'
__import__('pkg_resources').run_script('Conpot==0.5.1', 'conpot')
```
... fixes the warning.

Suggestions?

xandfury, 2018-08-02:
Tbh I don't like the dark arts of money patching - too many places where things can go wrong. [This is where the message originates from](https://github.com/mushorg/conpot/blob/783422a47a20d93e6a14b1b432e59872930fc03d/bin/conpot#L21). When you run `/usr/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/bin/conpot` monkey patching is the first thing we do, so no errors can be seen. I suspect this warning may have appeared during the build process. If that is the case, we have nothing to fear.

In any case, the wrapper suggest by you may patch libraries which we don't want to be patched (I am not sure about this). For example I always `import socket` from `gevent` rather than importing `socket` and patching it to be gevent compatible. So a higher level wrapper would even patch sockets where I intended to use convential sockets.

t3chn0m4g3, 2018-08-06:
@xandfury It's a runtime warning, does not happen during the build process.

xandfury, 2018-08-06:
@t3chn0m4g3 Any logs we can look at?

t3chn0m4g3, 2018-08-06:
@xandfury If I remember correctly the only messages I got or could retrieve from the log files are posted above.

xandfury, 2018-08-13:
@t3chn0m4g3 Can you try this with the latest release? See if the problem persists?

t3chn0m4g3, 2018-08-16:
@xandfury This is fixed.

closed 2018-08-16 by t3chn0m4g3