mushorg/conpot #365: Setup install error - Build docker image from source

omergunal, 2018-04-07:
I am getting this output: https://ghostbin.com/paste/hwajr

There is no problem via a pre-built image

xandfury, 2018-04-07:
For a quick fix, change the `Dockerfile` and [comment out the tests](https://github.com/mushorg/conpot/blob/d157229e4587188ad3d3af5dddcd71200713852d/Dockerfile#L24).

omergunal, 2018-04-12:
@xandfury 
my dockerfile: https://ibb.co/bA33U7
this time i got this error.
Error: https://ibb.co/iWZown

xandfury, 2018-04-12:
@omergunal The error is not about the template file. `Conpot` has been installed successfully. You need to supply a custom config file based on [this](https://github.com/mushorg/conpot/blob/master/conpot/testing.cfg).

If you are just testing out things, append `/bin/sh` to your above running command. Then start conpot manually using: `conpot -f --template default`. 

**Do not run the testing config in production however**. It would make you obvious to detect. The "errors" you get are deliberate, again to avoid having users run the testing config.

omergunal, 2018-04-13:
It worked, thanks

closed 2018-04-13 by omergunal