mushorg/conpot #368: increase the number of slaves

omergunal, 2018-04-27:
When I scan with nmap, I see 2 slaves.
[image]

how can i build 1 more slave?

xandfury, 2018-04-27:
Check your modbus template. [Example](https://github.com/mushorg/conpot/blob/master/conpot/templates/default/modbus/modbus.xml)

omergunal, 2018-04-27:
I create new slave in this field (https://github.com/mushorg/conpot/blob/master/conpot/templates/default/modbus/modbus.xml#L11) like `memoryModbusSlave0BlockA` but it was not enough

like that
`               <block name="memoryModbusSlave0BlockX">
                    <type>COILS</type>
                    <starting_address>500</starting_address>
                    <size>128</size>
                    <content>memoryModbusSlave0BlockA</content>
                </block> `

xandfury, 2018-04-27:
You need [`<slave id=""></slave>`](https://github.com/mushorg/conpot/blob/master/conpot/templates/default/modbus/modbus.xml#L70-L89)

omergunal, 2018-04-27:
I added this in modbux.xml, but nothing changed
[image]

closed 2018-04-27 by omergunal