mushorg/conpot #369: conpot: error: unrecognized arguments: -f

sereysethy, 2018-05-03:
Hi,

I think there is a small error in the README file, the option `-f` does not exist. I used a docker version.

```
# conpot -f --template default

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.5.1
  MushMush Foundation

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -f

```

xandfury, 2018-05-04:
Did you build the container from source or fetched one from dockerhub?

sereysethy, 2018-05-04:
I pulled it directly from dockerhub, I just want to test it first.

sereysethy, 2018-05-04:
I tried to build using `docker-compose`, the build failed when running coverage. Please check the log below:

Running env:
- OS: `Mac OS Sierra 10.13.4`
- Docker for Mac: `18.03.1-ce-mac65 (24312)`
- docker-compose: `1.21.1`
```
$ docker-compose build
Building conpot
Step 1/15 : FROM python:2
 ---> cb627aaf5f26
Step 2/15 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 120881b51d1a
Step 3/15 : RUN sed -i -e 's/main/main non-free contrib/g' /etc/apt/sources.list
 ---> Using cache
 ---> 9b9bc1cb7d53
Step 4/15 : RUN apt-get update -y -qq && apt-get install -y -qq         libmysqlclient-dev         libxslt1-dev &&     apt-get clean &&     rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*
 ---> Using cache
 ---> 27bcfdf3424f
Step 5/15 : COPY ./ /opt/conpot/
 ---> Using cache
 ---> 934c9e76e776
Step 6/15 : WORKDIR /opt/conpot
 ---> Using cache
 ---> 01dcf8cdc2fa
Step 7/15 : RUN pip install --no-cache-dir coverage
 ---> Using cache
 ---> 817aac771887
Step 8/15 : RUN pip install --no-cache-dir -r requirements.txt
 ---> Using cache
 ---> b2ed601b06d6
Step 9/15 : RUN coverage run --timid --source=conpot setup.py test
 ---> Running in 249913d8222c
running test
Searching for nose
Reading https://pypi.org/simple/nose/
Downloading https://files.pythonhosted.org/packages/99/4f/13fb671119e65c4dce97c60e67d3fd9e6f7f809f2b307e2611f4701205cb/nose-1.3.7-py2-none-any.whl#sha256=dadcddc0aefbf99eea214e0f1232b94f2fa9bd98fa8353711dacb112bfcbbb2a
Best match: nose 1.3.7
Processing nose-1.3.7-py2-none-any.whl
Installing nose-1.3.7-py2-none-any.whl to /opt/conpot/.eggs

Installed /opt/conpot/.eggs/nose-1.3.7-py2.7.egg
running egg_info
creating Conpot.egg-info
writing requirements to Conpot.egg-info/requires.txt
writing Conpot.egg-info/PKG-INFO
writing top-level names to Conpot.egg-info/top_level.txt
writing dependency_links to Conpot.egg-info/dependency_links.txt
writing manifest file 'Conpot.egg-info/SOURCES.txt'
reading manifest file 'Conpot.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Conpot.egg-info/SOURCES.txt'
running build_ext
WARNING:scapy.loading:Cannot read wireshark manuf database
WARNING:scapy.loading:Failed to execute tcpdump. Check it is installed and in the PATH
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)
ERROR:scapy.interactive:Loading module scapy.layers.netflow
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/scapy/main.py", line 121, in _load
    mod = importlib.import_module(module)
  File "/usr/local/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/usr/local/lib/python2.7/site-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/scapy/layers/netflow.py", line 86, in <module>
    class NetflowRecordV5(Packet):
  File "/usr/local/lib/python2.7/site-packages/scapy/layers/netflow.py", line 101, in NetflowRecordV5
    ByteEnumField("prot", IP_PROTOS["tcp"], IP_PROTOS),
  File "/usr/local/lib/python2.7/site-packages/scapy/dadict.py", line 38, in __getitem__
[... 96 lines cut ...]
Objective: Test if the spoofer is able to change MAC address
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/conpot/conpot/tests/test_mac_addr.py", line 38, in test_mac
    s = subprocess.Popen(["modprobe", "dummy"], stderr=subprocess.STDOUT, stdout=subprocess.PIPE)
  File "/usr/local/lib/python2.7/site-packages/gevent/subprocess.py", line 585, in __init__
    reraise(*exc_info)
  File "/usr/local/lib/python2.7/site-packages/gevent/subprocess.py", line 554, in __init__
    restore_signals, start_new_session)
  File "/usr/local/lib/python2.7/site-packages/gevent/subprocess.py", line 1312, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
-------------------- >> begin captured logging << --------------------
conpot.protocols.IEC104.IEC104_server: INFO: IEC104 Station disconnected. (2260826d-1850-42f9-9348-5836e62878b7)
--------------------- >> end captured logging << ---------------------

----------------------------------------------------------------------
Ran 44 tests in 42.038s

FAILED (errors=2, skipped=3)
Test failed: <unittest.runner.TextTestResult run=44 errors=2 failures=0>
Coverage.py warning: Trace function changed, measurement is likely wrong: <bound method PyTracer._trace of <PyTracer at 139713308371216: 4126 lines in 95 files>> (trace-changed)
error: Test failed: <unittest.runner.TextTestResult run=44 errors=2 failures=0>
ERROR: Service 'conpot' failed to build: The command '/bin/sh -c coverage run --timid --source=conpot setup.py test' returned a non-zero code: 1
```

xandfury, 2018-05-04:
Could you try again after [commenting out the tests](https://github.com/mushorg/conpot/blob/d157229e4587188ad3d3af5dddcd71200713852d/Dockerfile#L24)? See if that works for you

xandfury, 2018-05-04:
@sereysethy Like I said, conpot does create a logfile by default.  For something more docker related - [this should help](https://github.com/mushorg/conpot/issues/355).

xandfury, 2018-05-10:
@sereysethy 
Were you able to resolve your issue? 
I think we can close this.

sereysethy, 2018-05-11:
@xandfury as far as I can tell build is successful, but I havent actually analysed captured logs. Anyway will the test be fixed?

sereysethy, 2018-05-11:
well it works now since I added option `-f` to run conpot but it still checks for wireshark.

```
conpot_1  | WARNING:scapy.loading:Cannot read wireshark manuf database
conpot_1  | WARNING:scapy.loading:Failed to execute tcpdump. Check it is installed and in the PATH
conpot_1  | WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)
conpot_1  | ERROR:scapy.interactive:Loading module scapy.layers.netflow
conpot_1  | Traceback (most recent call last):
conpot_1  |   File "/usr/local/lib/python2.7/site-packages/scapy/main.py", line 121, in _load
conpot_1  |     mod = importlib.import_module(module)
conpot_1  |   File "/usr/local/lib/python2.7/importlib/__init__.py", line 37, in import_module
conpot_1  |     __import__(name)
conpot_1  |   File "/usr/local/lib/python2.7/site-packages/gevent/builtins.py", line 96, in __import__
conpot_1  |     result = _import(*args, **kwargs)
conpot_1  |   File "/usr/local/lib/python2.7/site-packages/scapy/layers/netflow.py", line 86, in <module>
conpot_1  |     class NetflowRecordV5(Packet):
conpot_1  |   File "/usr/local/lib/python2.7/site-packages/scapy/layers/netflow.py", line 101, in NetflowRecordV5
conpot_1  |     ByteEnumField("prot", IP_PROTOS["tcp"], IP_PROTOS),
conpot_1  |   File "/usr/local/lib/python2.7/site-packages/scapy/dadict.py", line 38, in __getitem__
conpot_1  |     return getattr(self, attr)
conpot_1  | AttributeError: DADict instance has no attribute 'tcp'
conpot_1  | 2018-05-11 12:26:55,618 --force option specified. Using testing configuration:
conpot_1  | 2018-05-11 12:26:55,620 Starting Conpot using template: /usr/local/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default
conpot_1  | 2018-05-11 12:26:55,620 Starting Conpot using configuration found in: /usr/local/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/testing.cfg
conpot_1  | 2018-05-11 12:26:55,674 Fetched 138.48.202.82 as external ip.
conpot_1  | 2018-05-11 12:26:55,678 Conpot modbus initialized
conpot_1  | 2018-05-11 12:26:55,680 Found and enabled ('modbus', <class 'conpot.protocols.modbus.modbus_server.ModbusServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,684 Conpot S7Comm initialized
conpot_1  | 2018-05-11 12:26:55,685 Found and enabled ('s7comm', <class 'conpot.protocols.s7comm.s7_server.S7Server'>) protocol.
conpot_1  | 2018-05-11 12:26:55,687 Found and enabled ('http', <class 'conpot.protocols.http.web_server.HTTPServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,689 Found and enabled ('snmp', <class 'conpot.protocols.snmp.snmp_server.SNMPServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,691 Conpot Bacnet initialized using the /usr/local/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/bacnet/bacnet.xml template.
conpot_1  | 2018-05-11 12:26:55,692 Found and enabled ('bacnet', <class 'conpot.protocols.bacnet.bacnet_server.BacnetServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,693 IPMI BMC initialized.
conpot_1  | 2018-05-11 12:26:55,694 Conpot IPMI initialized using /usr/local/lib/python2.7/site-packages/Conpot-0.5.1-py2.7.egg/conpot/templates/default/ipmi/ipmi.xml template
conpot_1  | 2018-05-11 12:26:55,695 Found and enabled ('ipmi', <class 'conpot.protocols.ipmi.ipmi_server.IpmiServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,697 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
conpot_1  | 2018-05-11 12:26:55,698 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
conpot_1  | 2018-05-11 12:26:55,698 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
conpot_1  | 2018-05-11 12:26:55,700 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
conpot_1  | 2018-05-11 12:26:55,700 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
conpot_1  | 2018-05-11 12:26:55,701 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
conpot_1  | 2018-05-11 12:26:55,702 Found and enabled ('enip', <class 'conpot.protocols.enip.enip_server.EnipServer'>) protocol.
conpot_1  | 2018-05-11 12:26:55,703 No proxy template found. Service will remain unconfigured/stopped.
conpot_1  | 2018-05-11 12:26:55,704 Modbus server started on: ('0.0.0.0', 502)
conpot_1  | 2018-05-11 12:26:55,705 S7Comm server started on: ('0.0.0.0', 102)
conpot_1  | 2018-05-11 12:26:55,705 HTTP server started on: ('0.0.0.0', 80)
conpot_1  | 2018-05-11 12:26:55,814 SNMP server started on: ('0.0.0.0', 161)
conpot_1  | 2018-05-11 12:26:55,815 Bacnet server started on: ('0.0.0.0', 47808)
conpot_1  | 2018-05-11 12:26:55,816 IPMI server started on: ('0.0.0.0', 623)
conpot_1  | 2018-05-11 12:26:55,817 handle server PID [    1] running on ('0.0.0.0', 44818)
conpot_1  | 2018-05-11 12:26:55,817 handle server PID [    1] responding to external done/disable signal in object 139801924020384
conpot_1  | 2018-05-11 12:27:00,674 Privileges dropped, running as "nobody:nogroup"
```

xandfury, 2018-05-12:
@sereysethy Warnings are not Conpot related. They are about `scapy`. Regarding `-f` option, it is intentional behaviour. It means you are using the testing config bundled with Conpot, which you shouldn't.

xandfury, 2018-06-12:
It seems that this issue is resolved (or longer relevant to you). If you need further assistance, feel free to re-open the issue again.

closed 2018-06-12 by xandfury