mushorg/conpot #370: hpfriends enable or log capture help!

sereysethy, 2018-05-04:
Hi everyone,

I am pretty new to the `conpot` and `hpfriends` log sharing system, but I was unable to find a very good place to get full information. The `http://hpfriends.honeycloud.net/` was not responding or I dont even know how to get an account to order to use this service.

I will be grateful if somebody can point me to a good direction as to where to start.

In other words, I just need a way to capture generated logs from `conpot`, which I think it is what the system is designed for.

xandfury, 2018-05-04:
`Conpot` does generate a logfile called `conpot.log` (if that works for you).  Ofcourse there are other ways to export logs besides `hpfeeds`. For example - [taxii](https://github.com/mushorg/conpot/blob/d157229e4587188ad3d3af5dddcd71200713852d/conpot/testing.cfg#L44). You can learn more about taxii [here](https://oasis-open.github.io/cti-documentation/)

I did google for more info about `hpfeeds`. Found good articles [here](https://heipei.github.io/2013/05/11/Using-hpfriends-the-social-data-sharing-platform/) and [here](https://heipei.github.io/2013/05/10/The-architecture-behind-hpfriends/).

sereysethy, 2018-05-04:
Yes but I am not sure yet if I need to run my own broker, because the http://hpfriends.honeycloud.net/ is currently down or it stops working already. So conpot supports taxii out of the box?

glaslos, 2018-05-08:
the hpfriends service has been discontinued. I'd recommend using the log file and some sort of log ingestion/aggregation. There are also some hints in the T-Pot or MHN projects on how to run your own hpfeeds broker.

closed 2018-05-08 by glaslos

sereysethy, 2018-05-09:
Thank you very much.