mushorg/conpot #372: Changing modbus registers in real time

angelferni90, 2018-05-15:
Hello guys, 

I would like to modify internally the value of modbus conpot registers (holding registers, coils) so the modbus clients that are reading, can see that the value of the holding registers is changing. 
I've seen that the the 'slave.py' and the 'modbus_block_databus_mediator.py' creates the registers blocks based on the file 'modbus.xml' and 'template.xml'. However, the values defined there for example in the file 'template.xml' we define using keys the values we want the modbus client to read:
<key name="myblock">
           <value type="value">[22,11,45,88,44,76,23,45,99]</value>
</key>

But those values are just the inicial value of the registers. What I would like is to run a script to modify them in real time, for example to make a temperature profile. 

Can somebody give me an idea how to start?

Thanks in advance.

Best regards.

cnlzxin, 2018-05-16:
socket ?

xandfury, 2018-05-16:
@cnlzxin Have you thought about interfacing an actual arduino based modbus device?
You can proxy conpot with that device - you might need to use ethernet shield but you can program Arduino for whatever values, even add a real temperature sensor.

If you want to dynamically modify register values, you need to change them in the databus.

angelferni90, 2018-05-28:
Hello and thanks @xandfury for your answer! 

Regarding the first option, I wouldn't like to add a physical device, I prefer to create/modify the values in the same machine where conpot is installed. 
So, to do this, I understand that the way to do it is changing them in the databus. 
I don't really know how to do it.. 

I see that there is a script called 'modbus_block_databus_mediator.py' that creates the registers in the databus based on the XML configuration file. But then, I don't know how to access to the databus to change the values. Do you have any idea how to do it?

Thanks so much in advance. 
Best regards

angelferni90, 2018-05-30:
Any help or ideas? @adepasquale @johnnykv @glaslos 

Thanks so much in advance!!

Vingaard, 2018-05-30:
@xandfury - just a input - when proxy interfacing with a real modbus devices, the internal logging goes haywire (as you need to specific what decoder to use in proxy.xml and currently there are no modbus decoder avaliable) .. could you clarified/ state the decoder that should be used) unless I have missed something in the latest development, Thanks for any input/knowledge

@angelferni90 - I am currently preparing several workshops on honeypots (one with conpot as 'base' lab) and I might take the challenge and write a small script who dynamic change the values..
Got a few ideas, and If I succeed, in making something stable - i would be be most happy to share this code with you and the rest of the Conpot community, when finish. 
Kind Regards /Mikael, Honeypot.dk

angelferni90, 2018-05-30:
Hello @Vingaard ! 
Thanks for your answer!! That would be very nice when you finish it!
I would like to start also doing this by myself, but the problem is that I don't know how to access to the data bus to change the values. 
Have you thought about how to access to the databus from the python script? Could you help me at list to start doing something? 

Thanks so much in advance. 
For other contributors,, any idea will be welcomed. 
thanks!!

xandfury, 2018-06-01:
Hey @angelferni90! Did you solve your issue?
Hacking though/changing databus values while running Conpot may not be a trivial task. Since both conpot and your script are different processes, you are looking at a mechanism that exposes the databus in some way. 

A local server would work. Your script would connect to that server acting as client and change relevant values. While this can fit somewhere with #250 - for now, let us leave this for the community to implement. :-)

xandfury, 2018-06-01:
@Vingaard You are right about that. I have been thinking about extending the proxy module for a while now. Users can just plug their own decoders. Like in your case, a modbus decoder might not difficult to implement - ofcourse we should ship with own set of decoders. But IMO the proxy module is one the most powerful features of Conpot. This would be a big help in realizing it's full potential ;-)
Daniel (@creolis) what do you think?

glaslos, 2026-09-14:
## Using #635 for #372-style dynamic registers

PR #635 already wires the hard part: Modbus blocks are **shared databus lists**, and `PlcScanCycle` mutates them every scan so clients see live values. Build the temperature profile (or any profile) as another **engine**, not as a separate process poking Conpot.

### Pattern

1. Keep the `plc_modbus` layout: coils / discrete / holding / analog as databus lists, `PlcScanCycle` started via `type="function"`.
2. Add a `PlcEngine` (same `scan(image)` contract as `NativeLogicEngine`) that writes your profile into `image.holding[...]` (and coils/analog if needed) each cycle.
3. Register it in `_make_engine` (e.g. `"profile"`) and select it from `template.xml` `param`.
4. Drive the profile from constructor args or a small template asset (CSV/JSON of setpoints over time)—still loaded **inside** the Conpot process.

Because the mediator reads the same lists Modbus serves, no extra hooks are required.

### Minimal sketch

```python
class TemperatureProfileEngine:
    def __init__(self, points, hold_index=0):
        self.points = points  # [(t_ms, value), ...]
        self.hold_index = hold_index
        self.t = 0

    def scan(self, image):
        # pick setpoint for self.t, write image.holding[self.hold_index]
        self.t += 50  # match scan_ms
```

Point `engine_name` at this class in the template; leave Modbus XML block names unchanged.

### What this gives vs what it doesn’t

- **Gives:** real-time register changes visible to Modbus readers, same path as the native plant PoC.
- **Doesn’t:** an external script in another process. For that, later add a localhost control API that mutates the same lists the scan cycle already holds—or have the engine read setpoints from a file the script updates.

**Bottom line:** treat #635 as the I/O image + scan loop; implement #372 as a custom `PlcEngine` that writes the profile into that image each tick.

closed 2026-09-20 by glaslos