mushorg/conpot #376: README.md Easy install using Docker via prebuilt image syntax problem

jamiebuening, 2018-06-19:
I just followed the Easy install using Docker via a pre-built image instructions in the README.md.  It looks like the `-f` is no longer a valid command line argument.

```
# conpot -f --template default

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.5.1
  MushMush Foundation

usage: conpot [-h] [-t TEMPLATE] [-c config.cfg] [-l LOGFILE] [-a RAW_MIB]
              [-m MIBPATHS] [-v]
conpot: error: unrecognized arguments: -f
```

Does this argument just need to be removed from step 2?

xandfury, 2018-06-20:
It seems that unfortunately last build was an year ago. The `-f` option was introduced later than that. Since I don't have read/write  access there, I can't re-initiate the build.

Please try building the image from source.

xandfury, 2018-06-21:
Closing in favor of #378

closed 2018-06-21 by xandfury