mushorg/conpot #378: Update honeynet/conpot docker image

xandfury, 2018-06-21:
Last build of [honeynet/conpot](https://hub.docker.com/r/honeynet/conpot/builds/) happend over an year ago. Conpot has changed since them. Someone with `honeynet` org access needs to trigger the rebuild 

or

create a `mushorg` account on dockerhub, where we can push the latest code for automated builds

adepasquale, 2018-06-21:
@xandfury if you have a Docker hub account I can add you to the team, if @glaslos is OK with that.

Meanwhile I'll go and trigger a manual build.

xandfury, 2018-06-21:
I do have a [Docker Hub account](https://hub.docker.com/r/xandfury). :-)

adepasquale, 2018-06-21:
Upon a second look, we already have the hooks in place to trigger builds after each push to mushorg/conpot master branch. The build is failing, though (even the one I triggered manually):

```
Build failed: The command '/bin/sh -c apt-get update -y -qq && apt-get install -y -qq libmysqlclient-dev libxslt1-dev && apt-get clean && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*' returned a non-zero code: 100
```
```
[91mE: Package 'libmysqlclient-dev' has no installation candidate
```

Maybe we need to check our base image and Dockerfile are aligned with Docker Hub?

xandfury, 2018-06-21:
@adepasquale Replacing `libmysqlclient-dev` with `default-libmysqlclient-dev` worked for me :-) 
 
The build would run fine but you may get errors due to failing tests. These tests would include : 
- test_mac (change in MAC address)
- test_mysqllogger
- ImportError (No module named Crypto.Cipher) - IPMI server
- ImportError (No module named wsgi) - test_ext_ip_util

During py3 migration and upon discussion with Daniel, we agreed to skip first 2. Latter 2 can be fixed by [installing pycrypto](https://github.com/xandfury/conpot/blob/2ac3568d3140b2023d26e8835b7c21028fa732cf/Dockerfile#L22) and changing imports from [`from gevent.wsgi import WSGIServer`](https://github.com/mushorg/conpot/blob/d157229e4587188ad3d3af5dddcd71200713852d/conpot/tests/test_ext_ip_util.py#L24) to `from gevent.pywsgi import WSGIServer`
 
Here is the link the [successful build](https://hub.docker.com/r/xandfury/conpot/builds/b3vtpzxyb32u9zzccrnyel8/) and the associated [Dockerfile](https://github.com/xandfury/conpot/blob/test-docker/Dockerfile). 

These changes have already been made for py3 branch ;-)

adepasquale, 2018-06-23:
Thanks for your pull request, @xandfury. The build has been successful!

https://hub.docker.com/r/honeynet/conpot/builds/

I have also re-enabled automated builds after pushes on `master`

closed 2018-06-23 by adepasquale