mushorg/conpot #383: Document ICS/Conpot related research

xandfury, 2018-07-09:
After discussion with Daniel (@creolis), I think it would be a good idea to document Conpot related research in the project wiki. This could include - 
- Tutorials on how to write a good template. Since wiki is ***community driven***, we can discuss myraid of devices (like #271). IMO as long as we share just links/images we can keep ourselves distant from any legal twilight. 
- Conpot use case scenarios, deployment strategies.
- Response to various attacks. Tools for attacking ICS devices. 
- All kinds of ICS/Conpot related research, links to papers etc.
- And finally what and what not to expect from Conpot. 

Wiki is also under version control. So we don't have a lot hassle maintaining it. Also for a variety of reasons I don't think there is a place for the above points in the documentation. Opinion/suggestions? Point 2 is debatable on whether it should be in docs or wiki.

glaslos, 2018-07-12:
I think all of them are good candidates for the wiki

aman566, 2019-01-16:
Please assign it to me @glaslos  I would like to work. As by working on it I will know more about conpot. It would a help to me.

creolis, 2019-01-16:
:) you don't need to be assigned - you can always start working on an issue and keep us posted about your progress in order to show that there's activity on the matter.

north-echo, 2026-04-08:
I've drafted initial wiki content covering several of the points mentioned here. The pages are up on my fork's wiki for review:

- [Production Deployment Guide](https://github.com/north-echo/conpot/wiki/Production-Deployment-Guide) — Rootless Podman, container security hardening, firewall zone isolation, SELinux, audit rules, health monitoring
- [Writing Custom Templates](https://github.com/north-echo/conpot/wiki/Writing-Custom-Templates) — Template structure, realistic device fingerprints, cross-protocol consistency, testing methodology
- [Attack Patterns and Analysis](https://github.com/north-echo/conpot/wiki/Attack-Patterns-and-Analysis) — Three-tier traffic model (background noise → protocol scanning → targeted interaction), SSH attack patterns, bot vs. human indicators
- [ICS Honeypot Research](https://github.com/north-echo/conpot/wiki/ICS-Honeypot-Research) — Curated papers and related projects

Based on running a Conpot deployment in production. Happy to iterate on feedback before these go to the upstream wiki.

closed 2026-09-20 by glaslos