mushorg/conpot #393: Add type hints to code

xandfury, 2018-08-12:
Now that we have migrated to python 3.5, we should also look at adding type hints to our code (some 3.5+ sweet stuff right there)
Depending on editor you use, this could save us from making errors related to python's dynamic typing.

Example: 
https://github.com/mushorg/conpot/blob/2f48b3e44f0618fb30f60bd55bdb2e03487feaea/conpot/core/__init__.py#L48-L54

llucifer97, 2018-10-15:
hi @xandfury  I would like to take up this issue

xandfury, 2018-10-16:
@llucifer97 That's very nice to hear :slightly_smiling_face: 

I would suggest you to start reading about types and type hints as a primer. Next, go through conpot's source code and add those hints to objects - cross verify if necessary :wink: . Finally, add mypy or pydantic to tox, integrating type checking to our CI.

[This](https://stackoverflow.com/a/32558710/2873994) should get you started :wink:.

In not so later stages, we would to auto-generate our documentation using `automodule` feature. Something on lines of -
```
.. automodule:: conpot.core
    :members: fs
```

aman566, 2019-01-14:
@xandfury I can work on this issue. If no one has started then I can do. Please assign it to me .

glaslos, 2019-01-14:
@aman566 feel free to work on this :+1:

sh-ran, 2026-11-06:
Is this still open to pick up on? @glaslos