mushorg/conpot #395: Can't start conpot with default file system

t3chn0m4g3, 2018-08-16:
When starting 0.6.0 I am getting the following error:
```
/tmp $ conpot --template default --logfile /tmp/log.log --temp_dir /tmp -c /etc/conpot/conpot.cfg
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

2018-08-16 10:31:31,647 Config file found!
2018-08-16 10:31:31,649 Starting Conpot using template: /usr/lib/python3.6/site-packages/conpot/templates/default
2018-08-16 10:31:31,650 Starting Conpot using configuration found in: /etc/conpot/conpot.cfg
ERROR:root:Can't start conpot with default file system
2018-08-16 10:31:31,654 Can't start conpot with default file system
```

Maybe you can shed some light on this?

creolis, 2018-08-16:
Hey.
Since we are often criticized because of the fact that dozens of Conpot instances are using the default templates in the wild and therefore are easy to identify, using default values must be "enforced".

So everytime you want to start conpot without changing defaults (default template, default virtual file system, etc.), conpot will be harsh to you :) 

For local testing purposes, to tinker around and of course for unit testing, you can force conpot to start nevertheless. Use "-f" to tell it that default values are the new black <3 

Hope that helps!

t3chn0m4g3, 2018-08-16:
@creolis Thanks.

It seems the default path values for FTP are causing this:

```
  File "/usr/bin/conpot", line 455, in <module>
    main()
  File "/usr/bin/conpot", line 375, in main
    server = server_class(protocol_template, root_template_directory, args)
  File "/usr/lib/python3.6/site-packages/conpot/core/protocol_wrapper.py", line 27, in __init__
    self.wrapped = cls(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/conpot/protocols/ftp/ftp_server.py", line 225, in __init__
    self.handler.config = FTPConfig(self.template)
  File "/usr/lib/python3.6/site-packages/conpot/protocols/ftp/ftp_server.py", line 103, in __init__
    self._init_fs()        # Initialize FTP file system.
  File "/usr/lib/python3.6/site-packages/conpot/protocols/ftp/ftp_server.py", line 117, in _init_fs
    perms=self.default_perms)
  File "/usr/lib/python3.6/site-packages/conpot/core/__init__.py", line 55, in add_protocol
    return virtualFS.add_protocol(protocol_name, data_fs_subdir, vfs_dst_path, src_path, owner_uid, group_gid, perms)
  File "/usr/lib/python3.6/site-packages/conpot/core/virtual_fs.py", line 124, in add_protocol
    perms)
  File "/usr/lib/python3.6/site-packages/conpot/core/filesystem.py", line 810, in mount_fs
    raise fs.errors.DirectoryExpected('{} path does not exists'.format(path))
fs.errors.DirectoryExpected: path '/data/ftp path does not exists' should be a directory
```

Even with a changed `kamstrup_382` the VFS error is present:
```
conpot --template kamstrup_382 --logfile /tmp/log.log --temp_dir /tmp -c /etc/conpot/conpot.cfg
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

2018-08-16 12:06:08,732 Config file found!
2018-08-16 12:06:08,733 Starting Conpot using template: /usr/lib/python3.6/site-packages/conpot/templates/kamstrup_382
2018-08-16 12:06:08,734 Starting Conpot using configuration found in: /etc/conpot/conpot.cfg
ERROR:root:Can't start conpot with default file system
2018-08-16 12:06:08,747 Can't start conpot with default file system
```

Maybe I am not getting the feature, should it work right away or are some requirements necessary which need to be met?

t3chn0m4g3, 2018-08-16:
Just wanted to delete the docker container and got the following error as well. Maybe this is related:
```
docker rm $(docker ps -qa)
Error response from daemon: driver "overlay2" failed to remove root filesystem for ccee9714d4b437e174d4cc18d3267ee1b3b692962028557a1a1a8d6fefed1691: lstat /var/lib/docker/overlay2/0e76968baedd81b53d4f3334fa300f8d6aba19cf9f402aaa8790336e3dca4db8/diff/tmp/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6: file name too long
```

t3chn0m4g3, 2018-08-16:
I have been looking at the sample out:
```
2018-08-09 19:13:15,085 Initializing Virtual File System at ConpotTempFS/__conpot__ootc_k3j. Source specified : tar://conpot-0.6.0-py3.6/conpot/data.tar
2018-08-09 19:13:15,100 Please wait while the system copies all specified files
```
Do I need to specify the source somewhere?

t3chn0m4g3, 2018-08-16:
From the looks of it, the `default` values for `[virtual_file_system]` are the problem.
1. `data_fs_url` must be an absolute, existing path with r/w permissions in the file system
2. `fs_url` must be a `.tar` with a FS template as provided

Now this works for VFS, but now I have to fix the ENV interpolation:
```
[virtual_file_system]
data_fs_url = /tmp/
fs_url = tar:///usr/lib/python3.6/site-packages/conpot/data.tar
```

xandfury, 2018-08-16:
@t3chn0m4g3 Absolutely correct! But in your case, these default values should and must have been picked up by conpot. I think this has something to do with your ENV. If you can post the logs, I would be able to help you better.

t3chn0m4g3, 2018-08-16:
@xandfury This seems to be unrelated to ENVs, result was the same (even with #396 which brings #360 back) in place. As long as I do not use the `default` values it is working now. Except for `kamstrup`. Opening new issue for that.

xandfury, 2018-08-16:
Now that #396 is closed (thanks for that :smiley: ) maybe we can revisit this problem. TBH something like this
```
Error response from daemon: driver "overlay2" failed to remove root filesystem for ccee9714d4b437e174d4cc18d3267ee1b3b692962028557a1a1a8d6fefed1691: lstat /tmp/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6/__conpot__j9czgaq6
```
should and must never happen (looks like recursive directory creation). With all good intentions, we can't help you much without the detailed log file and the exact config (env) variables.

t3chn0m4g3, 2018-08-16:
Thank you @xandfury , but once I removed `default` from the VFS section I could not reproduce that (all the logs / tracebacks I got from conpot are posted in this issue. Replaced it with ENVs and everything works fine now (at least for my use cases).
You may close this, unless you want to investigate further on your side - happy to help 😄

xandfury, 2018-08-16:
@t3chn0m4g3 

My ₹0.02 (:wink:) is - you probably did not create an environment while installing conpot. This means that you are importing a lot of vairables where they might not be required. You can see this simply by
```
import os
os.environ
```
This may or may not work. But is generally discouraged in the python community. Instead you should always create a `virtualenv` whenever you plan to use ENV vars. Please try the following:
```
1. virtualenv --python=python3.5 conpot
2. source conpot/bin/activate
3. pip install conpot
4.  << your conpot command >>
```

Although I am aware that you are running `Conpot` in docker but still - we should always try to create the ENV. I started with #392 but couldn't find time to finish it yet. :smile:

xandfury, 2018-08-16:
I am closing this issue for now. But feel free to reopen if the problem persists.

closed 2018-08-16 by xandfury

t3chn0m4g3, 2018-08-16:
@xandfury Thanks.