mushorg/conpot #397: JSON does not log dst_port anymore

t3chn0m4g3, 2018-08-16:
JSON logs port as `null`:

Example FTP:
```
{"timestamp": "2018-08-16T15:40:08.504317", "sensorid": "conpot", "id": "69e26d22-2741-4760-8635-eb84e911ab09", "src_ip": "172.20.254.118", "src_port": 61815, "dst_ip": "84.154.38.33", "dst_port": null, "data_type": "ftp", "request": null, "response": null, "event_type": null}
{"timestamp": "2018-08-16T15:40:08.504317", "sensorid": "conpot", "id": "69e26d22-2741-4760-8635-eb84e911ab09", "src_ip": "172.20.254.118", "src_port": 61815, "dst_ip": "84.154.38.33", "dst_port": null, "data_type": "ftp", "request": null, "response": null, "event_type": null}
```

Example HTTP:
```
{"timestamp": "2018-08-16T15:36:34.171956", "sensorid": "conpot", "id": "11058a49-9e41-46a2-9086-f755bedff95f", "src_ip": "87.250.233.66", "src_port": 56113, "dst_ip": "84.154.38.33", "dst_port": null, "data_type": "http", "request": "('/index.php/view/control/ovcgi/page.php?pref=', [('Host', 'thebullocks.dynvpn.de'), ('Connection', 'Keep-Alive'), ('User-Agent', 'Mozilla/5.0 (compatible; YandexBot/3.0; +http://yandex.com/bots)'), ('From', 'support@search.yandex.ru'), ('Accept-Encoding', 'gzip,deflate'), ('Accept', '*/*')], None)", "response": "404", "event_type": null}
{"timestamp": "2018-08-16T15:38:11.732726", "sensorid": "conpot", "id": "63c9f0b9-7ec8-4d1e-9359-3a9e01584098", "src_ip": "5.255.251.17", "src_port": 55384, "dst_ip": "84.154.38.33", "dst_port": null, "data_type": "http", "request": "('/view/components/com_mtree/Savant2/public', [('Host', 'thebullocks.dynvpn.de'), ('Connection', 'Keep-Alive'), ('User-Agent', 'Mozilla/5.0 (compatible; YandexBot/3.0; +http://yandex.com/bots)'), ('From', 'support@search.yandex.ru'), ('Accept-Encoding', 'gzip,deflate'), ('Accept', '*/*')], None)", "response": "404", "event_type": null}
{"timestamp": "2018-08-16T15:38:56.094711", "sensorid": "conpot", "id": "02bdb083-82bf-4e9f-bf3a-1320df4d3b93", "src_ip": "172.20.254.118", "src_port": 61679, "dst_ip": "84.154.38.33", "dst_port": null, "data_type": "http", "request": "('/index.html', [('Host', 'lipstickenemy'), ('Connection', 'keep-alive'), ('Cache-Control', 'max-age=0'), ('Upgrade-Insecure-Requests', '1'), ('User-Agent', 'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/68.0.3440.106 Safari/537.36'), ('DNT', '1'), ('Accept', 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8'), ('Accept-Encoding', 'gzip, deflate'), ('Accept-Language', 'en-US,en;q=0.9,de-DE;q=0.8,de;q=0.7'), ('Cookie', 'sess_uuid=8910dcf1-4e30-4a9f-82a7-06314a26360b; sidebar_collapsed=false; path=/'), ('If-Modified-Since', 'Tue, 19 May 1993 09:00:00 GMT')], None)", "response": "200", "event_type": null}
```

xandfury, 2018-08-16:
`dst_ip` and `dst_port` logging was added recently in #344

t3chn0m4g3, 2018-08-16:
fixed the title accordingly.

vorband, 2018-08-17:
would be great if you could have a look at my comment regarding socket IP in #344, too. ;)

xandfury, 2018-08-17:
@vorband Thanks for the suggestion. Please feel free to send in a PR.

creolis, 2018-08-18:
@vorband if you could sent a PR that builds up on standa's merged code (I had no time to check in detail what went wrong yet) this would of course be appreciated :)

vorband, 2018-08-22:
I will come up with a PR as soon as I find the time.

t3chn0m4g3, 2018-08-22:
@xandfury @creolis Can you guys come up with a fix? I want to finish dockerizing by the end of this month.

vorband, 2018-08-22:
the missing dest_port/ip is due to #344 only logging protocols that use session_manager. So is my solution. FTP logging, as in the example above, is not using session_manager. At the moment I am more struggling with getting you the pull request with my changes for the session manager modifications. 
After this, I can try to understand how ftp is logged and add the desired info.

vorband, 2018-08-23:
ftp should now be fixed, as well as tftp in PR #400

t3chn0m4g3, 2018-08-23:
@vorband Thanks :bowtie: 
@creolis Will be testing tonight

t3chn0m4g3, 2018-08-24:
Testing was successful see #400

creolis, 2018-08-24:
Thanks for your quick intervention André! We appreciate it and I guess Marco owes you a beer ;-) (and we do too)

closed 2018-08-24 by creolis