mushorg/conpot #398: Kamstrup will not start up

t3chn0m4g3, 2018-08-16:
The reason seems to be related to a wrong path... After installation conpot expects `kamstrup_meter.xsd` in 
```
/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup_meter/kamstrup_meter.xsd
``` 
but the file is installed to 
```
/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup/meter_protocol/kamstrup_meter.xsd
```

Here the traceback...
```
conpot_kamstrup_382    | WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)
conpot_kamstrup_382    | 2018-08-16 15:44:03,377 Config file found!
conpot_kamstrup_382    | 2018-08-16 15:44:03,379 Starting Conpot using template: /usr/lib/python3.6/site-packages/conpot/templates/kamstrup_382
conpot_kamstrup_382    | 2018-08-16 15:44:03,379 Starting Conpot using configuration found in: /etc/conpot/conpot.cfg
conpot_kamstrup_382    | 2018-08-16 15:44:03,392 Serving tar:///usr/lib/python3.6/site-packages/conpot/data.tar as file system. File uploads will be kept at : /tmp
conpot_kamstrup_382    | 2018-08-16 15:44:03,393 Opening path /tmp for persistent storage of files.
conpot_kamstrup_382    | 2018-08-16 15:44:03,395 Initializing Virtual File System at /tmp/conpot/__conpot__7k4knjor. Source specified : tar:///usr/lib/python3.6/site-packages/conpot/data.tar
conpot_kamstrup_382    |  Please wait while the system copies all specified files
conpot_kamstrup_382    | 2018-08-16 15:44:03,458 Fetched 84.154.38.33 as external ip.
conpot_kamstrup_382    |
conpot_kamstrup_382    |                        _
conpot_kamstrup_382    |    ___ ___ ___ ___ ___| |_
conpot_kamstrup_382    |   |  _| . |   | . | . |  _|
conpot_kamstrup_382    |   |___|___|_|_|  _|___|_|
conpot_kamstrup_382    |               |_|
conpot_kamstrup_382    |
conpot_kamstrup_382    |   Version 0.6.0
conpot_kamstrup_382    |   MushMush Foundation
conpot_kamstrup_382    |
conpot_kamstrup_382    | Traceback (most recent call last):
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 455, in <module>
conpot_kamstrup_382    |     main()
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 363, in main
conpot_kamstrup_382    |     validate_template(protocol_template, xsd_file)
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 132, in validate_template
conpot_kamstrup_382    |     xml_schema = etree.parse(xsd_file)
conpot_kamstrup_382    |   File "src/lxml/etree.pyx", line 3426, in lxml.etree.parse
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1840, in lxml.etree._parseDocument
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1866, in lxml.etree._parseDocumentFromURL
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1770, in lxml.etree._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1163, in lxml.etree._BaseParser._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 601, in lxml.etree._ParserContext._handleParseResultDoc
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 711, in lxml.etree._handleParseResult
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 638, in lxml.etree._raiseParseError
conpot_kamstrup_382    | OSError: Error reading file '/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup_meter/kamstrup_meter.xsd': failed to load external entity "/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup_meter/kamstrup_meter.xsd"
conpot_kamstrup_382 exited with code 1
```

xandfury, 2018-08-16:
This is due to the change in protocol instance mappings:

https://github.com/mushorg/conpot/blob/2f48b3e44f0618fb30f60bd55bdb2e03487feaea/bin/conpot#L338-L340

I had changed directories for things to be more meaningful. We can do refactoring in `bin/conpot` or change directories like they were before.

t3chn0m4g3, 2018-08-22:
Thanks, tested the changes as proposed, but unfortunately it does not work.

```
conpot_kamstrup_382    |                        _
conpot_kamstrup_382    |    ___ ___ ___ ___ ___| |_
conpot_kamstrup_382    |   |  _| . |   | . | . |  _|
conpot_kamstrup_382    |   |___|___|_|_|  _|___|_|
conpot_kamstrup_382    |               |_|
conpot_kamstrup_382    | 
conpot_kamstrup_382    |   Version 0.6.0
conpot_kamstrup_382    |   MushMush Foundation
conpot_kamstrup_382    | 
conpot_kamstrup_382    | Traceback (most recent call last):
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 458, in <module>
conpot_kamstrup_382    |     main()
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 366, in main
conpot_kamstrup_382    |     validate_template(protocol_template, xsd_file)
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 132, in validate_template
conpot_kamstrup_382    |     xml_schema = etree.parse(xsd_file)
conpot_kamstrup_382    |   File "src/lxml/etree.pyx", line 3426, in lxml.etree.parse
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1840, in lxml.etree._parseDocument
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1866, in lxml.etree._parseDocumentFromURL
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1770, in lxml.etree._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1163, in lxml.etree._BaseParser._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 601, in lxml.etree._ParserContext._handleParseResultDoc
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 711, in lxml.etree._handleParseResult
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 638, in lxml.etree._raiseParseError
conpot_kamstrup_382    | OSError: Error reading file '/usr/lib/python3.6/site-packages/conpot/protocols/meter_protocol/meter_protocol.xsd': failed to load external entity "/usr/lib/python3.6/site-packages/conpot/protocols/meter_protocol/meter_protocol.xsd"
conpot_kamstrup_382 exited with code 1
```

xandfury, 2018-08-22:
@t3chn0m4g3 For the above code
```python3
if 'kamstrup' in protocol_name:
   kamstrup_dir, protocol_name = os.path.split((protocol_name.replace('_', '/')) + '_protocol')
   xsd_file = os.path.join(package_directory, 'protocols', kamstrup_dir, protocol_name, '{0}.xsd'.format(protocol_name))
else:
    xsd_file = os.path.join(package_directory, 'protocols', protocol_name, '{0}.xsd'.format(protocol_name)) 
```
can you debug for ` kamstrup_dir` and `protocol_name`? I don't have a machine to test `conpot` atm. 

Also I assume that you rebuilt conpot after making your changes :smile:

t3chn0m4g3, 2018-08-22:
@xandfury Traceback:
```
conpot_kamstrup_382    |                        _
conpot_kamstrup_382    |    ___ ___ ___ ___ ___| |_
conpot_kamstrup_382    |   |  _| . |   | . | . |  _|
conpot_kamstrup_382    |   |___|___|_|_|  _|___|_|
conpot_kamstrup_382    |               |_|
conpot_kamstrup_382    | 
conpot_kamstrup_382    |   Version 0.6.0
conpot_kamstrup_382    |   MushMush Foundation
conpot_kamstrup_382    | 
conpot_kamstrup_382    | Traceback (most recent call last):
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 459, in <module>
conpot_kamstrup_382    |     main()
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 367, in main
conpot_kamstrup_382    |     validate_template(protocol_template, xsd_file)
conpot_kamstrup_382    |   File "/usr/bin/conpot", line 132, in validate_template
conpot_kamstrup_382    |     xml_schema = etree.parse(xsd_file)
conpot_kamstrup_382    |   File "src/lxml/etree.pyx", line 3426, in lxml.etree.parse
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1840, in lxml.etree._parseDocument
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1866, in lxml.etree._parseDocumentFromURL
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1770, in lxml.etree._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 1163, in lxml.etree._BaseParser._parseDocFromFile
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 601, in lxml.etree._ParserContext._handleParseResultDoc
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 711, in lxml.etree._handleParseResult
conpot_kamstrup_382    |   File "src/lxml/parser.pxi", line 638, in lxml.etree._raiseParseError
conpot_kamstrup_382    | OSError: Error reading file '/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup/meter_protocol/meter_protocol.xsd': failed to load external entity "/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup/meter_protocol/meter_protocol.xsd"
conpot_kamstrup_382 exited with code 1
```

Folder structure after your the proposed changes:
```
/usr/lib/python3.6/site-packages/conpot/protocols/kamstrup $ ls -al -R
.:
total 28
drwxr-xr-x    5 root     root          4096 Aug 22 10:41 .
drwxr-xr-x   16 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root             0 Aug 22 10:37 __init__.py
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 __pycache__
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 management_protocol
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 meter_protocol
-rw-r--r--    1 root     root          4933 Aug 22 10:37 usage_simulator.py

./__pycache__:
total 20
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 .
drwxr-xr-x    5 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root           153 Aug 22 10:41 __init__.cpython-36.pyc
-rw-r--r--    1 root     root          4328 Aug 22 10:41 usage_simulator.cpython-36.pyc

./management_protocol:
total 56
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 .
drwxr-xr-x    5 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root             0 Aug 22 10:37 __init__.py
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 __pycache__
-rw-r--r--    1 root     root          2722 Aug 22 10:37 command_responder.py
-rw-r--r--    1 root     root         31286 Aug 22 10:37 commands.py
-rw-r--r--    1 root     root           464 Aug 22 10:37 kamstrup_management.xsd
-rw-r--r--    1 root     root          3931 Aug 22 10:37 kamstrup_management_server.py

./management_protocol/__pycache__:
total 44
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 .
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root           173 Aug 22 10:41 __init__.cpython-36.pyc
-rw-r--r--    1 root     root          1726 Aug 22 10:41 command_responder.cpython-36.pyc
-rw-r--r--    1 root     root         22554 Aug 22 10:41 commands.cpython-36.pyc
-rw-r--r--    1 root     root          3048 Aug 22 10:41 kamstrup_management_server.cpython-36.pyc

./meter_protocol:
total 60
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 .
drwxr-xr-x    5 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root             0 Aug 22 10:37 __init__.py
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 __pycache__
-rw-r--r--    1 root     root          2520 Aug 22 10:37 command_responder.py
-rw-r--r--    1 root     root          8300 Aug 22 10:37 decoder_382.py
-rw-r--r--    1 root     root          2355 Aug 22 10:37 kamstrup_constants.py
-rw-r--r--    1 root     root          1866 Aug 22 10:37 kamstrup_meter.xsd
-rw-r--r--    1 root     root          4980 Aug 22 10:37 kamstrup_server.py
-rw-r--r--    1 root     root          6124 Aug 22 10:37 messages.py
-rw-r--r--    1 root     root          1024 Aug 22 10:37 register.py
-rw-r--r--    1 root     root          3678 Aug 22 10:37 request_parser.py

./meter_protocol/__pycache__:
total 48
drwxr-xr-x    2 root     root          4096 Aug 22 10:41 .
drwxr-xr-x    3 root     root          4096 Aug 22 10:41 ..
-rw-r--r--    1 root     root           168 Aug 22 10:41 __init__.cpython-36.pyc
-rw-r--r--    1 root     root          1657 Aug 22 10:41 command_responder.cpython-36.pyc
-rw-r--r--    1 root     root          5271 Aug 22 10:41 decoder_382.cpython-36.pyc
-rw-r--r--    1 root     root          1754 Aug 22 10:41 kamstrup_constants.cpython-36.pyc
-rw-r--r--    1 root     root          3711 Aug 22 10:41 kamstrup_server.cpython-36.pyc
-rw-r--r--    1 root     root          5114 Aug 22 10:41 messages.cpython-36.pyc
-rw-r--r--    1 root     root           557 Aug 22 10:41 register.cpython-36.pyc
-rw-r--r--    1 root     root          2438 Aug 22 10:41 request_parser.cpython-36.pyc
```

creolis, 2018-08-29:
fixed in the course of #392

closed 2018-08-29 by creolis

t3chn0m4g3, 2018-08-30:
Thanks guys 😄