mushorg/conpot #403: docker permission denied

ntnychss, 2018-09-04:
Hi, I'm trying to run the dockerized  version of Conpot but I cant make it work.
I followed the _Via a pre-built image_ instructions.

here the error: 
[https://i.imgur.com/fXyE7ec.png](https://i.imgur.com/fXyE7ec.png)

I have tried also 'Build docker image from source' but got more or less the same kind of problem.

creolis, 2018-09-05:
The documentation is far from being accurate at the moment - sorry.
This only applies when you force conpot to run as root, which is not supported.

Please run conpot as non-privileged user. If you want to bind ports <=1024, 
use authbind or set linux capabilities :)

ntnychss, 2018-09-06:
@rajeshkalluri009  I solved it by installing it in a ubuntu 14.04 LTS virtual machine.
Installed it from pip ([guide](https://conpot.readthedocs.io/en/latest/installation/ubuntu.html)).
You will face error installing it. I purged all distutils and installed module with the version required by conpot. This is an iterative process you will face consecutive errors, just be patience, solve them one by one and it will work :)

creolis, 2018-09-06:
@rajeshkalluri009 You're sure you're not UID 0? (this also includes not using sudo btw.)

ntnychss, 2018-09-17:
@creolis if you run conpot as a non-root it will trigger "_### Permission Denied_" exceptions..

creolis, 2018-09-17:
May I ask you to provide a more specific output, rather than "permission denied exceptions" ? Related to filesystem calls? related to socket interaction? I can only assume that this is related to the way socket handling has changed - but you would help us to set that straight if you could post a decent log output.

Oh, btw.: with all due respekt, please try to not post screenshots. Not only are they harder to interpret and screenreader unfriendly - it's also very bad for archival reasons (when we pull out this issue in 12 months for reference and your IMGUR uploads are gone, there's no way telling what led us to our conclusions).

Conpot is a CLI application - please provide text output whenever you can  <3

xandfury, 2018-09-24:
I just checked. It seems that the conpot installation is done with the `root` user by default - something we should change. 

So the actual conpot python script is run by `root` as well and when prvis are dropped there are associated permission issues. Instead the Dockerfile should create a non-root user first - and use that user to install conpot, consequently running it.  

So the way I look it, it's a bug in our Dockerfile :smiley:  Thanks, we'll look into it :wink:

aaravkumar20, 2018-09-25:
to solve permission denied in conpot 0.6 use this easy step way:
**ubuntu 18.04**
**-install conpot with a non-root user**

-add this source to your repository & update:
deb http://archive.ubuntu.com/ubuntu $(lsb_release -sc) main universe multiverse restricted
-install git & python3-pip & python3-setuptools & python3.6-dev
-dont forget install stix-validator & cybox
-git clone https://github.com/mushorg/conpot.git
-cd conpot
-sudo python3 setup.py install
-conpot -f -t

-after installation , if you have permissionerror like this :
"PermissionError: [Errno 13] Permission denied : 'directory_address' "
-grant recursive permission (read & write) to this directory with : 
sudo chmod -R a+rw 'directory address'

run conpot with : 
conpot -f -t default

good lucky!

Pouick, 2018-09-26:
user@honey:~/conpot$ docker build -t conpot .
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.29/build?buildargs=%7B%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile&labels=%7B%7D&memory=0&memswap=0&networkmode=default&rm=1&shmsize=0&t=conpot&target=&ulimits=null: dial unix /var/run/docker.sock: connect: permission denied

xandfury, 2018-09-27:
@wintermute101 I highly doubt that. Please go ahead 😄

wintermute101, 2018-09-27:
@xandfury Copy that.

wintermute101, 2018-09-28:
Hi I've been working on this for a while.
I've moved docker related stuff to sub directory and now it is in depended. Maybe move it to different repository?
I did not create a pull request as I want to discuss it and it does not work. Maybe you can take a look.  There are some permission errors when adding tftp protocol.
Code: [docker-fixes branch](https://github.com/wintermute101/conpot/tree/docker-fixes)

Error:
```
WARNING:scapy.loading:Cannot read wireshark manuf database
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)
WARNING:root:--force option specified. Using testing configuration
2018-09-28 11:47:32,313 --force option specified. Using testing configuration
2018-09-28 11:47:32,314 Starting Conpot using template: /usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/templates/default
2018-09-28 11:47:32,315 Starting Conpot using configuration found in: /usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/testing.cfg
WARNING:conpot.core.virtual_fs:Using default FS path. tar:///usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/data.tar
2018-09-28 11:47:32,320 Using default FS path. tar:///usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/data.tar
2018-09-28 11:47:32,323 Initializing Virtual File System at /tmp/__conpot__b7_7zddg. Source specified : tar:///usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/data.tar
 Please wait while the system copies all specified files
2018-09-28 11:47:32,405 Fetched 91.192.15.73 as external ip.
2018-09-28 11:47:32,412 Conpot modbus initialized
2018-09-28 11:47:32,413 Found and enabled ('modbus', <conpot.protocols.modbus.modbus_server.ModbusServer object at 0x7f5b7c6add68>) protocol.
2018-09-28 11:47:32,417 Conpot S7Comm initialized
2018-09-28 11:47:32,417 Found and enabled ('s7comm', <conpot.protocols.s7comm.s7_server.S7Server object at 0x7f5b7c768be0>) protocol.
2018-09-28 11:47:32,419 Found and enabled ('http', <conpot.protocols.http.web_server.HTTPServer object at 0x7f5b7c768ba8>) protocol.
2018-09-28 11:47:32,420 Found and enabled ('snmp', <conpot.protocols.snmp.snmp_server.SNMPServer object at 0x7f5b7c768ef0>) protocol.
2018-09-28 11:47:32,422 Conpot Bacnet initialized using the /usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/templates/default/bacnet/bacnet.xml template.
2018-09-28 11:47:32,422 Found and enabled ('bacnet', <conpot.protocols.bacnet.bacnet_server.BacnetServer object at 0x7f5b7c7861d0>) protocol.
2018-09-28 11:47:32,424 IPMI BMC initialized.
2018-09-28 11:47:32,425 Conpot IPMI initialized using /usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/templates/default/ipmi/ipmi.xml template
2018-09-28 11:47:32,426 Found and enabled ('ipmi', <conpot.protocols.ipmi.ipmi_server.IpmiServer object at 0x7f5b7c7865c0>) protocol.
2018-09-28 11:47:32,428 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-09-28 11:47:32,429 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2018-09-28 11:47:32,430 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-09-28 11:47:32,431 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
2018-09-28 11:47:32,431 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2018-09-28 11:47:32,431 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2018-09-28 11:47:32,432 Found and enabled ('enip', <conpot.protocols.enip.enip_server.EnipServer object at 0x7f5b7c786a90>) protocol.
2018-09-28 11:47:32,434 Creating persistent data store for protocol: ftp
2018-09-28 11:47:32,439 FTP Serving File System at /data/ftp/ in vfs. FTP data_fs sub directory: /ftp
2018-09-28 11:47:32,446 Found and enabled ('ftp', <conpot.protocols.ftp.ftp_server.FTPServer object at 0x7f5b7c6c3a20>) protocol.
2018-09-28 11:47:32,447 Creating persistent data store for protocol: tftp
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/site-packages/fs/osfs.py", line 306, in makedir
    os.mkdir(sys_path, mode)
PermissionError: [Errno 13] Permission denied: b'/usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/tests/data/data_temp_fs/tftp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 4, in <module>
    __import__('pkg_resources').run_script('conpot==0.6.0', 'conpot')
  File "/usr/local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 661, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1441, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/EGG-INFO/scripts/conpot", line 461, in <module>
    main()
  File "/usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/EGG-INFO/scripts/conpot", line 381, in main
    server = server_class(protocol_template, root_template_directory, args)
  File "/usr/local/lib/python3.5/site-packages/conpot-0.6.0-py3.5.egg/conpot/core/protocol_wrapper.py", line 27, in __init__
    self.wrapped = cls(*args, **kwargs)
[... cut ...]
  File "/usr/local/lib/python3.5/site-packages/fs/osfs.py", line 314, in makedir
    return self.opendir(_path)
  File "/usr/local/lib/python3.5/site-packages/fs/error_tools.py", line 86, in __exit__
    reraise(fserror, fserror(self._path, exc=exc_value), traceback)
  File "/usr/local/lib/python3.5/site-packages/six.py", line 692, in reraise
    raise value.with_traceback(tb)
  File "/usr/local/lib/python3.5/site-packages/fs/osfs.py", line 306, in makedir
    os.mkdir(sys_path, mode)
fs.errors.PermissionDenied: permission denied
```

xandfury, 2018-10-01:
@wintermute101 Sorry for the late reply. 

- I can only suggest you to debug for permissions by appending `/bin/bash` or `/bin/sh` to your docker run command. Changing users is always tricky with docker when you need to work with files. :smile:
 
- I am fine with having a separate directory for docker, I don't think there should be a separate repo just for docker. (Things get harder to maintain, keep track of etc.) 

- Why don't you try the installation with the `conpot` user itself? We need to run conpot from non-root user after all :wink: IIRC there is a `--user` option with pip.

In any case, thanks for your effort :+1:  ; would love to see a working docker build of conpot

wintermute101, 2018-10-01:
Thanks for suggestions.  What I think this is not related to docker itself, anyway I'll debug it next week. I'm off grid for a week.

xandfury, 2018-10-09:
If you run the tests, a tftp client uploads and downloads a test file. As long they pass, I doubt there is implementation related issue. Logs would be nice to see nevertheless :wink:

wintermute101, 2018-10-09:
Yes that is strange to me also as tests do pass. After installing conpot tftp does not start. Logs with failing tftp due to permissions are above. I will try to figure it out but I will need to go deeper into tftp code.

wintermute101, 2018-10-11:
It works now in docker. I used supervisor to start and control conpot and it runs process not as root but as conpot user. I think supervisor is nice choice if you want to start other processes in docker like SIEM log forwarder, STIX or MHN hpfeeds broker.

closed 2018-10-29 by glaslos

handsome113530, 2019-11-28:
> 该文档目前还远远不够准确-抱歉。
> 仅当您强制conpot以root身份运行时才适用，这不受支持。
> 
> 请以非特权用户身份运行conpot。如果要绑定<= 1024的端口，请
> 使用authbind或设置linux功能:)
yingjun@yingjun-virtual-machine:/opt/conpot$ sudo conpot
WARNING:scapy.loading:Cannot read wireshark manuf database
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

CRITICAL:root:Can't start conpot with root. Please ref user docs for more info.
2019-11-28 11:30:49,444 Can't start conpot with root. Please ref user docs for more info.

 **Hello, I still don't understand how to solve this problem. What command should I input**

handsome113530, 2019-11-28:
> The documentation is far from being accurate at the moment - sorry.
> This only applies when you force conpot to run as root, which is not supported.
> 
> Please run conpot as non-privileged user. If you want to bind ports <=1024,
> use authbind or set linux capabilities :)
yingjun@yingjun-virtual-machine:/opt/conpot$ sudo conpot
WARNING:scapy.loading:Cannot read wireshark manuf database
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

CRITICAL:root:Can't start conpot with root. Please ref user docs for more info.
2019-11-28 11:30:49,444 Can't start conpot with root. Please ref user docs for more info.

**Hello, I still don't understand how to solve this problem. What command should I input**

PR #416 (2018-10-29): Docker fixes
fixes #403, moved docker related stuff to subdir, updated README accordingly