mushorg/conpot #405: conpot installing error

sepantakntu, 2018-09-16:
hello!
 i installed conpot ==0.6.0, after i run conpot -f --template default below error appear? how can help me to solve this problem!
Traceback (most recent call last):
  File "/usr/local/bin/conpot", line 4, in <module>
    __import__('pkg_resources').run_script('conpot==0.6.0', 'conpot')
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 750, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1524, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/conpot-0.6.0-py2.7.egg/EGG-INFO/scripts/conpot", line 32, in <module>
    import conpot.core as conpot_core
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/conpot-0.6.0-py2.7.egg/conpot/core/__init__.py", line 48
    def add_protocol(protocol_name: str,
                                  ^
SyntaxError: invalid syntax

xandfury, 2018-09-17:
You are trying to run Conpot with python2.7. You need to run it with python 3.5

closed 2018-09-17 by xandfury

mechgt, 2018-09-17:
I was facing this exact same issue, and was not able to resolve this with python 3.5, but python **3.6** worked.  

Python 3.5 (on CentOS 7) was returning:
`ImportError: cannot import name 'Text'`
This was in reference to 
`from typing import Optional, Union, Text, Any, List` in `filesystem.py`

... something to do with the timing of when Text was added to typing in python 3.5

creolis, 2018-09-17:
:) In general: The current release of conpot (starting with Conpot 0.6) is strictly relying on Python3.
If for one or the other reason you need to stick to Python2, the last compatible release is Conpot 0.5.2 (a.k.a: Python2 retirement release)