mushorg/conpot #406: docker permission denied not root.

wlfrag, 2018-09-24:
so after running 
Install Docker
Run docker pull honeynet/conpot
Run docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh

I proceeded to run conpot -f --template default as a non-root user, I created a new user and didn't add him to the sudoers group, and I ran the command and it shows me permission denied.

https://imgur.com/a/NXooLfS

aaravkumar20, 2018-09-24:
-Add the following line to the source : sudo nano /etc/apt/sources.list
deb http://archive.ubuntu.com/ubuntu $(lsb_release -sc) main universe multiverse restricted

-Install this dependencies:
python3-setuptools & python3-pip
-then install this package with pip3:
stix-validator & cybox & upgrade pip

-Add the following line to the source : sudo nano /etc/apt/sources.list
ppa:jonathonf/python-3.6
install python3.6-dev & git
git clone https://github.com/mushorg/conpot.git
cd conpot
python setup.py install

wlfrag, 2018-09-24:
hello aarav, so after i add the line to the source
"-Add the following line to the source : sudo nano /etc/apt/sources.list
deb http://archive.ubuntu.com/ubuntu $(lsb_release -sc) main universe multiverse restricted"

I need to sudo apt-get update to get the new source list?

wlfrag, 2018-09-24:
@aaravkumar20 
followed ur guide from a fresh ubuntu 16.04 installation and i got this error message

https://imgur.com/a/djP3LfR

kummmar, 2018-09-24:
I am experiencing the same issue. We are not allowed to run docker build without sudo. When we used sudo to run the program, the program told us to not run as root and quit. We are currently stuck in a loop. No root no docker. Got root no conpot

xandfury, 2018-09-24:
Hi there.
Thanks for reporting. This seems to be a duplicate of #403. Please join the discussion there if needed be :smile:

closed 2018-09-24 by xandfury