mushorg/conpot #408: docker-compose issue

wlfrag, 2018-09-24:
on fresh ubuntu16
[image]

xandfury, 2018-09-24:
Should be fixed with #403

wintermute101, 2018-09-25:
I had similar issue it was actually problem with tftp tests failing.
I fixed it by following patch:
`diff --git a/conpot/protocols/tftp/tftp_handler.py b/conpot/protocols/tftp/tftp_handler.py
index 44a6239..c52d8f9 100644
--- a/conpot/protocols/tftp/tftp_handler.py
+++ b/conpot/protocols/tftp/tftp_handler.py
@@ -60,10 +60,10 @@ class TFTPServerState(TFTPState):
             # Return same state, we're still waiting for valid traffic.
             return self
         logger.debug("Requested filename is %s", pkt.filename)
-        if pkt.filename.startswith(self.context.root.encode()):
+        if pkt.filename.startswith(self.context.root):
             full_path = pkt.filename
         else:
-            full_path = os.path.join(self.context.root, pkt.filename.decode().lstrip('/'))
+            full_path = os.path.join(self.context.root, pkt.filename.lstrip('/'))
         try:
             logger.info("Full path of file to be uploaded is {}".format(full_path))
             self.full_path = full_path`

wintermute101, 2018-09-25:
@xandfury this is docker-compose build issue with tests failing I do not think this is related to #403

xandfury, 2018-09-26:
Right. Can you send in a pull request that would fix this? But before that can you post the conpot logs (in text - not image) ? For future reference and for us to know the context 😄

wintermute101, 2018-09-26:
Pull request done #411

creolis, 2018-09-26:
Fixed by #411, thanks guys.

closed 2018-09-26 by creolis