mushorg/conpot #413: Could not find config file

santidediego, 2018-10-03:
When I run` sudo conpot -t default` I get the error:

_Could not find config file!
Use -f option to try the test configuration__

I would like to run Conpot without using the test configuration, how should I use a configuration file? In [this doc](https://conpot.readthedocs.io/en/latest/installation/configuration.html) is written a sample config file, but no information about how to use it.

xandfury, 2018-10-06:
What is the complete command you are running? The correct command should look like `conpot -c /my_path/my_config.cfg -t default`

santidediego, 2018-10-08:
It´s fine, no worries, making the docs is not the most thrilling thing... 

The problem is: the conpot is not reading the path correctly with `-c` option, it searches in a fixed location, so if you have a config file in the current directory, for example, conpot won't be able to read it. I made a pull request which I think fixes this issue.

Regarding to sudo, I use this issue to ask another question, if you don't mind: how can I put (for example) port 80 for HTTP without running the tool as sudo?

Thanks for your previous answer, I'm trying to understand how the tool works 😄

xandfury, 2018-10-09:
I just checked and it worked fine for me. What command are you using? Can you also post your config file? Also post the detailed logs reported by conpot.

As for the second question, If you want to bind ports <=1024, use `authbind` or set linux capabilities

santidediego, 2018-10-10:
I repetead the experiment and I run `conpot -c conpot.cfg -t default` and I got:
`Could not find config file!` It´s trying to get the file in: 
`<path>/venv/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/EGG-INFO/scripts/conpot` this is, in a fixed directory inside the virtualenv.

However, if you write the full path as you did: `conpot -c /my_path/my_config.cfg -t my_custom_template`, it works. Anyway, I think is better the conpot is able to recognise the current directory, so the user can decide how he wants his config file read, but your choice, you can accept the pull request or just close the issue.

Thanks for the `authbind` idea, I´ll try it.

xandfury, 2018-10-17:
> I repetead the experiment and I ran `conpot -c conpot.cfg -t default` and I got:
`Could not find config file!` It´s trying to get the file in: 
`<path>/venv/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/EGG-INFO/scripts/conpot` this is, in a fixed directory inside the virtualenv.

Correct! Path is relative to the conpot executable. I have no issues in making this change. Please see review at #414

xandfury, 2018-10-25:
I am closing issue this since #414 in now closed. Thanks for the effort anyway :slightly_smiling_face:

closed 2018-10-25 by xandfury