mushorg/conpot #419: error with Build docker image from source

Pouick, 2018-11-15:
OS : debian Stretch 9

after : 

> sudo docker build -t conpot .

`error: command 'gcc' failed with exit status 1

      ----------------------------------------
  Command "/usr/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-28jufq96/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-c_emnmpd/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-a8ra7fd1 --compile" failed with error code 1 in /tmp/pip-install-28jufq96/cffi/

  ----------------------------------------
Command "/usr/bin/python3.6 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-a8ra7fd1 --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools>=18.5 wheel "cffi>=1.7,!=1.11.3; python_implementation != 'PyPy'"" failed with error code 1 in None
The command '/bin/sh -c pip3 install --user --no-cache-dir -r requirements.txt' returned a non-zero code: 1
`

xandfury, 2018-11-15:
Maybe this is related to #418. Can you please check and confirm?

Pouick, 2018-11-19:
I do not understand what to do .. I installed the dependencies by hand but always this error

`compilation terminated.
      error: command 'gcc' failed with exit status 1

      ----------------------------------------
  Command "/usr/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/                                                                                                              '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pi                                                                                                              --compile" failed with error code 1 in /tmp/pip-install-0a9n99oo/cffi/

  ----------------------------------------
Command "/usr/bin/python3.6 -m pip install --ignore-installed --no-user --prefix                                                                                                             i.org/simple -- setuptools>=18.5 wheel "cffi>=1.7,!=1.11.3; python_implementatio
The command '/bin/sh -c pip3 install --user --no-cache-dir -r requirements.txt'`

thank you

Pouick, 2018-11-21:
it's ok
thank you !

closed 2018-11-21 by Pouick