mushorg/conpot #420: docker run error

Pouick, 2018-11-21:
**Method : Build docker image from source**
`docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp -p 47808:47808 -p 623:6230 -p 21:2121 -p 69:6969/udp -p 44818:44818 --network=bridge conpot`

_

> Unable to find image 'conpot:latest' locally
> docker: Error response from daemon: repository conpot not found: does not exist or no pull access.
> See 'docker run --help'.

_

pietroferretti, 2018-11-21:
You probably first need to build the Docker image with `docker build -t conpot .`.

From the other issue you opened (#419) I guess you didn't manage to build it successfully. Have you tried the fix suggested in the issue #418? That worked for me.

Pouick, 2018-11-22:
Hi pietroferretti
the issue 419 is now closed
finally, I managed to build the image (I did not notice any error)
only when I run the last command I get the message above

I try to build the image again in case.

thank you

Pouick, 2018-11-22:
I try to build the image again and this time, it blocks:
```
_Step 22/32: COPY --from = conpot-builder --chown = conpot: conpot /home/conpot/.local/ /home/conpot/.local/
Unknown flag: chown_
```

the first build lasted 15 minutes and seemed to work ..

Pouick, 2018-12-05:
I just tried again on a fresh installation (the 15th attempt) and always an error ...
Someone really managed to operate by following one of the 3 methods ??

_```
docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp -p 47808:47808/udp -p 623:6230/udp -p 21:2121 -p 69:6969/udp -p 44818:44818 --network=bridge conpot
Unable to find image 'conpot:latest' locally
docker: Error response from daemon: pull access denied for conpot, repository does not exist or may require 'docker login'.
See 'docker run --help'.
```_

glaslos, 2018-12-05:
You probably want `honeynet/conpot:latest`

pietroferretti, 2018-12-05:
I did manage to run it.
If your problem is still `Unknown flag: chown`, a quick search on google reveals that the flag is only supported on Docker CE >= 17.09. Have you tried updating your Docker installation to the latest version?
Anyway IMHO there doesn't seem to be any problem with this project's Dockerfile.

aman566, 2019-01-16:
docker: Error response from daemon: driver failed programming external connectivity on endpoint compassionate_engelbart (73c99186010112867d98b416e77d89174224af1f3478574815026f44fb2ade30): Error starting userland proxy: listen tcp 0.0.0.0:80: bind: address already in use.
Makefile:6: recipe for target 'run-docker' failed
make: *** [run-docker] Error 125

@glaslos  please help me with this issue.