mushorg/conpot #430: Testing conpot default template

Hadi-94, 2019-02-03:
I've installed Conpot (The default template) and set it up using "virtualenv" documentation that is available.
I want to test it to check if it is actually working and detecting attacks from a different computer on the same network. I've already tested it using nmap (for checking http and ftp ports), but I'm more interested in Modbus and S7comm protocols.
Since I'm still new at this can anyone help me out in suggesting some tools that can help ?
Thanks in advance..

10rule, 2019-02-04:
Modbus and S7 both have nmap scripts

https://nmap.org/nsedoc/scripts/modbus-discover.html
https://nmap.org/nsedoc/scripts/s7-info.html

Hadi-94, 2019-02-16:
Thanks for the help !! ur suggestion was helpful @10rule

Hadi-94, 2019-02-18:
There is an issue every time I run the nmap code with the script provided for s7-comm,  the outpot script doesn't look like how it should be. 

First I ran the script against port 102 and i got this result:
https://imgur.com/a/t8tdhcz

Second, I ran it against port 1020 and I got this result:
https://imgur.com/a/aXpx7GI

there was no details about the s7-comm in the nmap script output. In addition the Honeypot didn't detect the attack at all, as if it never happened. 

Pls help because Conpot is a major part in my FYP and I'm really lost.

Hadi-94, 2019-02-24:
Update on this issue: 
I've found out that Conpot stopped using ports <=1024, and sometimes the port numbers were entered wrong. 
However, the issue of not having the same expected results still remains, and I think if i tried to bind Conpot using authbind as it was suggested in one of the issues before it might help in getting information from S7comm protocol and Modbus Protocol.

closed 2019-02-24 by Hadi-94