mushorg/conpot #431: Conpot not found

MiniPierre, 2019-02-04:
When trying to launch conpot via the `docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh` command, the shell does appear but the `conpot -f --template default` command gave me the following output : `/bin/sh: conpot: not found`

xandfury, 2019-02-04:
Can you try again replacing `/bin/sh` with `/bin/bash`?

10rule, 2019-02-04:
OK, there are actually 2 parts to your question:

1. How do I run conpot from the shell inside the container
2. How do I run conpot from docker

Answer to the first question:
Inside the container: The conpot executable is actually in 
`/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/bin/conpot`

Likewise, the default template is actually in.
`/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default`

So using the full path from inside the container, run:
```
/home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/bin/c
onpot -f --template /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0
-py3.6.egg/conpot/templates/default
```

2. To run the default template from docker, just remove the /bin/sh part.
`sudo docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot`
The reason this works is because Dockerfile runs default config by default.

MiniPierre, 2019-02-05:
Thank you for your quick answer, it did work well with your Docker command. Just a quick precision, the HTTP server is running on port 8800, so the command should be `sudo docker run -it -p 80:8800 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot`

creolis, 2019-02-06:
The docs are unfortunately not reflecting the current state of development. We moved to using ports > 1024 since it has two main advantages:

1st: You don't need to run conpot as root in order to bind to them. If you WANT to bind to ports < 1024, which is usually the case as soon as you're going into production, you can use several techniques in order to allow conpot to do so (setcap/CAP_NET_BIND_SERVICE, authbind, ..).

2nd: People that spawn a conpot instance for testing usually don't modify templates and waste their IPs. As long as they do not really go into production, it's better to have them not using the "good" ports in order to avoid being flagged as a honeypot in under 5 minutes :)

closed 2019-07-08 by xandfury

cr3wcu7, 2020-01-22:
Good afternoon. Faced the problem above, need help.

```
**root@r00t-VirtualBox:~# sudo docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp --network=bridge honeynet/conpot**
WARNING:scapy.runtime:No route found for IPv6 destination :: (no default route?)

                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|

  Version 0.6.0
  MushMush Foundation

WARNING:root:--force option specified. Using testing configuration
2020-01-22 15:47:56,160 --force option specified. Using testing configuration
2020-01-22 15:47:56,163 Starting Conpot using template: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default
2020-01-22 15:47:56,163 Starting Conpot using configuration found in: /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/testing.cfg
WARNING:conpot.core.virtual_fs:Using default FS path. tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
2020-01-22 15:47:56,173 Using default FS path. tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
2020-01-22 15:47:56,176 Initializing Virtual File System at /tmp/__conpot__p6kft9ii. Source specified : tar:///home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/data.tar
 Please wait while the system copies all specified files
2020-01-22 15:47:56,502 Fetched 92.49.180.58 as external ip.
2020-01-22 15:47:56,508 Conpot modbus initialized
2020-01-22 15:47:56,508 Found and enabled modbus protocol.
2020-01-22 15:47:56,511 Conpot S7Comm initialized
2020-01-22 15:47:56,512 Found and enabled s7comm protocol.
2020-01-22 15:47:56,514 Found and enabled http protocol.
2020-01-22 15:47:56,515 Found and enabled snmp protocol.
2020-01-22 15:47:56,517 Conpot Bacnet initialized using the /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/bacnet/bacnet.xml template.
2020-01-22 15:47:56,518 Found and enabled bacnet protocol.
2020-01-22 15:47:56,520 IPMI BMC initialized.
2020-01-22 15:47:56,520 Conpot IPMI initialized using /home/conpot/.local/lib/python3.6/site-packages/conpot-0.6.0-py3.6.egg/conpot/templates/default/ipmi/ipmi.xml template
2020-01-22 15:47:56,520 Found and enabled ipmi protocol.
2020-01-22 15:47:56,523 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:47:56,524 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2020-01-22 15:47:56,524 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:47:56,526 Class    22/0x0016, Instance   1, Attribute     3 <== [{'class': 22}, {'instance': 1}, {'attribute': 3}]
2020-01-22 15:47:56,526 Class    22/0x0016, Instance   1, Attribute     1 <== [{'class': 22}, {'instance': 1}, {'attribute': 1}]
2020-01-22 15:47:56,527 Class    22/0x0016, Instance   1, Attribute     2 <== [{'class': 22}, {'instance': 1}, {'attribute': 2}]
2020-01-22 15:47:56,528 Found and enabled enip protocol.
2020-01-22 15:47:56,532 Creating persistent data store for protocol: ftp
2020-01-22 15:47:56,546 FTP Serving File System at /data/ftp/ in vfs. FTP data_fs sub directory: /ftp
2020-01-22 15:47:56,573 Found and enabled ftp protocol.
2020-01-22 15:47:56,574 Creating persistent data store for protocol: tftp
2020-01-22 15:47:56,588 TFTP Serving File System at /data/tftp/ in vfs. TFTP data_fs sub directory: /tftp
2020-01-22 15:47:56,606 Found and enabled tftp protocol.
2020-01-22 15:47:56,606 No proxy template found. Service will remain unconfigured/stopped.
2020-01-22 15:47:56,607 Modbus server started on: ('0.0.0.0', 5020)
2020-01-22 15:47:56,608 S7Comm server started on: ('0.0.0.0', 10201)
2020-01-22 15:47:56,608 HTTP server started on: ('0.0.0.0', 8800)
2020-01-22 15:47:57,371 SNMP server started on: ('0.0.0.0', 16100)
2020-01-22 15:47:57,377 Bacnet server started on: ('0.0.0.0', 47808)
2020-01-22 15:47:57,378 IPMI server started on: ('0.0.0.0', 6230)
2020-01-22 15:47:57,380 handle server PID [    1] running on ('0.0.0.0', 44818)
2020-01-22 15:47:57,381 handle server PID [    1] responding to external done/disable signal in object 139888602675240
2020-01-22 15:47:57,382 FTP server started on: ('0.0.0.0', 2121)
2020-01-22 15:47:57,384 Starting TFTP server at ('0.0.0.0', 6969)
```

hAbd0u, 2020-05-15:
@cr3wcu7 Try this command, it has the correct ports:
```
sudo docker run -it -p 80:8800 -p 102:10201 -p 502:5020 -p 161:16100/udp --network=bridge honeynet/conpot
```