mushorg/conpot #433: Virtualenv and HMI

Hadi-94, 2019-02-16:
I'm running conpot using virtualenv documentation that is available. I've understood also from the documentation that usually if I want to check the default HMI that conpot has, I should open my browser and type in My IP address. However, the HMI page doesn't show up, instead it shows me a page says "Unable to connect". 
Can I know how to solve this issue ? Is it a problem because I'm running conpot using virtualenv instead of docker ? or it doesn't matter ?

xandfury, 2019-02-16:
You need to check what port conpot’s HTTP server is running on. Then point your browser at `http://ip_address:port`

Hadi-94, 2019-02-16:
Ahaa.. Thanks for the help

Hadi-94, 2019-02-16:
Should I do the same thing if I want to open the HMI from different device? @xandfury

closed 2019-02-16 by Hadi-94