mushorg/conpot #435: Deploying Conpot and

Hadi-94, 2019-02-18:
I have some questions regarding Conpot Deployment .. Hopefully anyone in the community can help me out.
I've ran conpot for 1 day using virtual environment, and there was no interaction with it from strange IP address (attacks). keeping in mind that I didn't adjust anything in the Simulation Python files as I'm afraid to mess it up.

Does running the conpot in the virtual enviroment means that the conpot is deployed online and can detect attacks from other IPs? If not then what is the missing step that can help in doing that?

Another question is how can I Increase the probability of having a high numbers of attacks ?

10rule, 2019-03-01:
> I've ran conpot for 1 day using virtual environment, and there was no interaction with it 

This is perfectly normal. Honeypots are meant to be trip wires, they only get triggered when there is an intruder inside your network.

>Does running the conpot in the virtual enviroment means that the conpot is deployed online and can detect attacks from other IPs?

The default way of deploying conpot is to bridge host's network to the docker container. (That's what `--network=bridge` is for). Conpot takes on the IP of your host machine.

An easy sanity check to see conpot is running is to do a modbus scan.
`nmap --script modbus-discover.nse --script-args='modbus-discover.aggressive=true' -p 502 1.2.3.4`
(Replace 1.2.3.4 with your conpot machine's IP address)

>Another question is how can I Increase the probability of having a high numbers of attacks ? How can I make conpot visible to the attackers ? I'v read online that using Shodan search engine is not a good idea as conpot have static configurations. So what should I do instead ?

Industrial networks is traditionally air gapped or behind a firewall. There shouldn't be a high number of attacks, because even 1 attack means your control network has been compromised.

Hadi-94, 2019-03-03:
>This is perfectly normal. Honeypots are meant to be trip wires, they only get triggered when there is an intruder inside your network.
> Industrial networks is traditionally air gapped or behind a firewall. There shouldn't be a high number of attacks, because even 1 attack means your control network has been compromised.

The thing is my research focus more on how attackers interact with SCADA Systems. So what I'm trying to do is that I'm deploying Conpot with no protection or firewalls, hoping to collect some information about attackers and how do they behave when there such system that is not fully protected. I know that Conpot is a low interaction Honeypot and its interaction is quite basic. What I'm hoping for is to collect as much attacks as possible ... any advises ?!

10rule, 2019-03-08:
>What I'm hoping for is to collect as much attacks as possible ... any advises ?!

Never done this myself, just brainstorming here. You might want to host your conpot online. Then register a bait website that connect back to it. This simulates an industrial control system that was unwittingly connected to the internet. For the bait web page, copy a simple HMI interface. It shouldn't be too hard since HMI looks like they were designed in the 90s.

>Another question, when I deploy the Honeypot using docker, the log's file location will be the same ?! on my computer ?!

By default, conpot logs to 2 places, stdout and conpot.log.

When running inside a docker container, docker daemon will aggregate all stdout into its log.
After starting up the docker container,

- Type `docker logs` to see the all the logs
- You can also try `docker logs --follow` to continuously see new logs

To see conpot.log inside the container is a bit more complicated.
You first start the container, docker will tell you a hash,that's the container's ID.
You use `docker exec -it ID /bin/sh` to connect into the container, in the default dir, you should see a conpot.log file (if not, use find `find / -name conpot.log`)

Hadi-94, 2019-03-09:
Alright .. this is really helpful . . @10rule Thanks for your advice 😁

closed 2019-03-09 by Hadi-94

Hadi-94, 2019-03-16:
> Never done this myself, just brainstorming here. You might want to host your conpot online. Then register a bait website that connect back to it. This simulates an industrial control system that was [...]

I don't know how usually HMIs looks like, I've been trying to look online and copy one, but mine looked really basic and not that interactive.
If u have a resource/website that might help me out regarding this, It'll be much appreciated.