mushorg/conpot #436: Authbind with Conpot

Hadi-94, 2019-02-24:
I've been trying to use Authbind tool with conpot to connect HTTP web server with port 80. 

I've configured Authbind tool in the root using these set of commands:

```
    touch /etc/authbind/byport/80
    chown $[user]:$[user] /etc/authbind/byport/80
    chmod 755 /etc/authbind/byport/80
```

After exiting the root, I ran the following command to make the server run on port 80:

`    authbind python "conpot/lib/python3.6/site-packages/conpot/protocols/http/web_server.py"`

The error that I'm getting is:

> ImportError: No module named conpot.protocols.http.command_responder 

Is the command that I gave right or wrong ? and how do I solve this issue?

creolis, 2019-02-24:
>> Is the command that I gave right or wrong ? and how do I solve this issue?

It's wrong :) 

Please do not start an individual protocol module - 
you start conpot as a whole :) 

Please consult the readme on how to start conpot, then extend that using authbind.

closed 2019-02-24 by creolis

Hadi-94, 2019-02-25:
Hello .. 
I did start conpot as a whole using virtualenv tutorial and I still couldn't figure it out .. Can I know the exact command for connecting authbind with conpot ?
your help would be appreciated ..

Hadi-94, 2019-02-25:
I've redefined the authbind tool to give access to the non-root user for these protocols.

https://imgur.com/edit?deletehash=OFI00U6FOKnUNOf&album_id=fJW4WF1

Then I tried to run the following command `authbind --deep conpot -f -t default`
yet still conpot can't seem to be using ports <=1024
I know that I'm missing something, if anyone has an idea pls help.. 
@creolis @xandfury

Hadi-94, 2019-02-26:
Never mind .. I'v already figured it out .. Thanks anyways

xandfury, 2019-02-27:
@Hadi-94 You can always share your feedback/experience for others who stumble upon this post 🙂

creolis, 2019-02-27:
> Never mind .. I'v already figured it out .. Thanks anyways

Sorry for the late replies, we had some tough days due to final GSoC preparations etc. -
but as for your final note: Please don't do that :) 

As @xandfury noted ... this is like finding yourself crawling through the desert on the verge of death and stumbling upon a super complicated water tap contraption which is surrounded by skeletons and a note that reads: "never mind, I figured it out", followed by a single trail of footsteps leading away from it .. (someone skilled should do a meme with that) ..

[image]

Hadi-94, 2019-02-27:
@creolis @xandfury I was going to share it with u, I just had a long day full of lectures so I didn't have the time to write it down 😅

However, these are the steps that I took to solve this issue: 

1- I've downloaded Authbind tool, and reconfigured it to use all ports <=1024.
For the ports that is < 512, u can reconfigure it as these commands in the root (repeat the same command for different ports, but change the port number):

`touch /etc/authbind/byport/80
    chown $[user]:$[user] /etc/authbind/byport/80
    chmod 755 /etc/authbind/byport/80`

for ports that is between 512 -1023, u have to adjust the commands to look like this:

`touch '/etc/authbind/byport/!623'
    chown $[user]:$[user] '/etc/authbind/byport/!623'
    chmod 755 '/etc/authbind/byport/!623'`

2- Adjust the template files for each of the protocols that will use these ports, by changing the "port" variable.

3-  Run the the following command:

`authbind conpot -f -t default`

Hopefully this is clear enough. and good luck for u guys while going through the GSoC program