mushorg/conpot #437: Conpot Freeze while Nmap Scann

Hadi-94, 2019-02-26:
Hello... 
I've been trying to use this command to scan Conpot and get information from it:
`nmap -A -Pn -p1-1000 172.20.10.2`

The output of this scan will come back as in the photo shown bellow:

[image]

The Conpot's output console will stop in the middle of the scan and the HMI webpage wouldn't open, so I'm assuming that Conpot itself will freeze.
This is the logs of Conpot provided in the file.
[Conpot's log](https://github.com/mushorg/conpot/files/2904902/Conpot.s.log)

Is this issue of Conpot's freezing is normal as a result of the scan that I performed ? and how can I avoid that from happening again ?!

10rule, 2019-03-01:
> Is this issue of Conpot freezing is normal as a result of the scan that I performed ? and how can I avoid that from happening again ?!

Hi,
I just started using Conpot a few month ago, just trying to help here.

From what I can see, you were scanning port 1 to 1000.
Port 102 is S7 port commonly used by PLCs.
The S7Server tried to service the request. 
While parsing the packet, the COTP parser threw an Exception (because it was trying to unpack data that wasn't there).
But the S7Server has no catch block to handle the exception.
So the exception traveled up outside the main server loop. 
And that's why your conpot stopped.

The easiest fix (not necessary the most correct), is to stop throwing that exception.
In file cotp.py line 44:
Replace 
    `raise ParseException('s7comm', 'malformed packet header structure')`
With 
    `return`

Hope that helps.

Hadi-94, 2019-03-03:
Hey .. 
I've tried your advice of replacing the line from 
`raise ParseException('s7comm', 'malformed packet header structure')`
to
`return` 
Here is Conpot's log while the scan is being performed:

[Conpot (2).log](https://github.com/mushorg/conpot/files/2923105/Conpot.2.log)

It didn't work, and Conpot still freezes. Moreover, there are some new errors that appeared, which I think is due to the fact that I removed the exception. 😥

What I know about COTP is that the connection between the source and the and receiver should be created so that later on, S7comm protocol can set up its own connection on top of it. What I also Know is that the COTP packets will only contain Acknowledgement and Sequence numbers, so that the connection should be established, so I think it is normal to have empty packets that doesn't contain data.

I've been trying different nmap scans directed to specific ports, I've tried to scan port 102 alone with the same scrip and there was no problem with Conpot, it didn't crash nor freeze at all, and it worked well.

On the other hand, when I scan port 502, which is used by Modbus protocol, Conpot freezes after a while and I had to kill it as usual. Here is the scan that I've been doing a long with Conpot's Log while the scan being performed. 

[conpot 3.log](https://github.com/mushorg/conpot/files/2923153/conpot.3.log)

So because of this I think the problem of Conpot freezing is because there is an error in the Modbus Protocol simulation. It might be a bit of a reach over here but that's what I'm noticing from my tests. 

@10rule thanks for your help and I really appreciate your insight. Pls do share any other ideas that you have that might help in solving this issue

Hadi-94, 2019-03-09:
@xandfury @creolis Any advice on this issue ?!

xandfury, 2019-03-11:
The exception has to be handled in the protocol s7 protocol implementation itself. :slightly_smiling_face:

Hadi-94, 2019-03-11:
@xandfury Are u sure that this is an issue of an S7 protocol ?!
When I directed the scan to port 102 only, Conpot didn't freeze at all. However, when I directed the scan towards port 502 (Modbus protocol), Conpot Freezes and doesn't detect attacks or listen to ports anymore.

10rule, 2019-03-11:
Sorry for the delay, had a busy week.

I just had a look at `Conpot (2).log`.
>conpot/protocols/s7comm/s7_server.py", line 92, in handle 
>`if cotp_base_packet.tpdu_type == 0xe0:`
>AttributeError: '`NoneType`' object has no attribute `'tpdu_type`'

Here is line 89 to 92:
>  tpkt_packet = TPKT().parse(cleanse_byte_string(data))
>  cotp_base_packet = COTP_BASE_packet().parse(tpkt_packet.payload)
>   if cotp_base_packet.tpdu_type == 0xe0:

A quick summery of this code (since I am assuming you are not too familiar with python),
First, we take the incoming packet, call TPKT parse on that and get back tpkt packet
Second we take the tpkt packet, call COPT parse() on that and get back a cotp packet
Third, we examine the type field of the cotp packet to see if it is equal hex e0.

Using my last fix, COTP parse will return a null packet, there is no tpdu_type in that null,
so the if statement threw another exception
and that exception traveled outside of the main server loop.

So quick and dirty fix v2.0,
wrap these two lines 
>  tpkt_packet = TPKT().parse(cleanse_byte_string(data))
>  cotp_base_packet = COTP_BASE_packet().parse(tpkt_packet.payload)
inside a a try except block.
Something like this
>try:
>    tpkt_packet = TPKT().parse(cleanse_byte_string(data))
>    cotp_base_packet = COTP_BASE_packet().parse(tpkt_packet.payload)
>except Exception as e:
>    break

What this code does is catch any exception from either of those two parse calls, and break which skips this iteration of the loop.

Hope that helps.

PS: In case you are wondering why nmap is causing this failure. The thing you got to understand about nmap is that it is an awesome tool but it is also really buggy around the edges. Sometimes the packets it constructs are not strictly valid, sometimes it fails for no reason. With nmap results, you got to assume it is 80% right.

PPS: I wish I can write up a proper fix to this. I am really resource constrained at this point. My apologies to the devs of this awesome project.

Hadi-94, 2019-03-16:
Hey 10Rule ..

This is my first time that I'm working an open source project written in a professional way (I'm still  an undergraduate student with a limited experience in coding), and I'm lost since there are some stuff that is new to me.

However, regarding the issue ... 
I've understood your explanation regarding COTP packet and what happens to it in s7_server.py file. I've tried to do your suggestion (v2.0 alone) in changing these lines only:

> wrap these two lines
>
>
> inside a a try except block.
> Something like this

This is the log file after performing the attack: 
`nmap -A -Pn -p1-1000 1.2.3.4`

[Conpot (4).log](https://github.com/mushorg/conpot/files/2973579/Conpot.4.log)

The error that was showing, which is this:

> conpot/protocols/s7comm/s7_server.py", line 92, in handle
> if cotp_base_packet.tpdu_type == 0xe0:
> AttributeError: 'NoneType' object has no attribute 'tpdu_type'

has disappeared, **yet conpot still freezes** at this line:

> 2019-03-16 15:05:10,732 Received unknown COTP TPDU before handshake: 32

I've took the initiative to turn off the modbus protocol from the configurations file, and performed the same attack, it went smoothly with no errors but one.

These are conpot log files while the attack being performed. 

[conpot (5).log](https://github.com/mushorg/conpot/files/2973601/conpot.5.log)

The only error that it has been repeated over and over is this one:

>   File "/home/ubuntutest/conpot/lib/python3.6/site-packages/conpot/protocols/http/command_responder.py", line 518, in send_error
>     self.headers = self.MessageClass(self.rfile, 0)
> TypeError: __init__() takes from 1 to 2 positional arguments but 3 were given

Ps: I'm more and more convinced that what is making conpot freezes is a problem in the modbus protocol simulation. From which I saw while performing more tests on modbus protocol alone, is that when the attacks doesn't carry any modbus scripts, it makes conpot freezes as conpot will stuck in a loop until I shut it down. I've reached to this conclusion after performing several tests and watching the packets being transformed using WireShark.

I would love to upload the details and files if any of the conpot team is interested in investigating this issue.

Hadi-94, 2019-03-16:
> PS: In case you are wondering why nmap is causing this failure. The thing you got to understand about nmap is that it is an awesome tool but it is also really buggy around the edges. Sometimes the [...]

This is true and I've noticed this several times while conducting my tests using nmap, Some times there will be issues saying that there are some `struct.error` or `malformed packet header structure`, which i think it is effected by the nmap scripts that sometimes aren't well constructed.

bestrocker221, 2021-04-23:
as of today, the behavior is still the same at the latest commit. Any news on this?
thank you

Kevalz95, 2025-12-02:
Hey @bestrocker221 @Hadi-94 , I know that is quite an old issue but I've just found a solution for the Modbus protocol handler freezing with NMAP aggressive scans (`nmap -A` and / or `nmap -sV`).

Basically, after some debugging, I've found out that the problem relies in the handle() method of conpot/protocols/modbus/modbus_server.py.

The main problem relies in the following while loop that downloads the Modbus TCP request from the connecting socket

```
_, _, length = struct.unpack(">HHH", request[:6])
while len(request) < (length + 6):
    try:
        new_byte = sock.recv(1)
        request += new_byte
    except Exception:
        break
```

It seems that, for some NMAP modbus requests, such while loop results into an infinite loop freezing the whole server and spiking the CPU usage to 100%!

More deeply, the `new_byte = sock.recv(1)` line is not considering the case where the sock.recv() doesn't receive any NEW byte and results with the while condition `len(request) < (length + 6)` never been achieved.

I've implemented and tested a fix for such issue by introducing a new variable called `previous_request` that, as the name say, it just stores the N-1 byte parsed by the socket and, if the bytes is the same for the following N+1000 received bytes, it just break the loop.

```
previous_request = None
stall_counter = 0
while len(request) < (length + 6):
    try:
        if previous_request and len(previous_request) == len(request):
            stall_counter += 1
            if stall_counter >= 1000:
                logger.info("Modbus client provided data {} but invalid.".format(session.id))
                session.add_event({"type": "CONNECTION_TERMINATED"})
                break
        new_byte = sock.recv(1)
        request += new_byte
        previous_request = request
    except Exception:
        break
```

I've tested such implementation against a NMAP Modbus scan (that was previously freezing the whole server) and it seems to work pretty well.

[image]

Hope it helps :)

glaslos, 2026-09-14:
Fixed in https://github.com/mushorg/conpot/pull/623

closed 2026-09-14 by glaslos