mushorg/conpot #438: Virtualization tool

Hadi-94, 2019-03-15:
What type of a virtualization tool does conpot use ?!
If it doesn’t use any, what type of a tool should I use to count and analyze the results with ?!

xandfury, 2019-03-15:
Conpot doesn't support any virtualization tools out of the box. I would recommend to checkout `T-pot` that comes with `conpot` pre-installed. It offers a really nice [kibana dashboard](https://github.com/dtag-dev-sec/tpotce#kibana) :slightly_smiling_face:

closed 2019-03-15 by xandfury

Hadi-94, 2019-03-15:
Does the copy of conpot that they have can be adjusted as the one here ?! Meaning that can I change the HMI interface and the basic configurations that conpot holds ?!

xandfury, 2019-03-16:
Yes. You should be able to. Contact the devs there

Hadi-94, 2019-03-17:
alright ... Thanks