mushorg/conpot #440: getting error during enip port scanning

prashantvidja, 2019-03-20:
Hello,

I have deploy conpot docker image. But when I scanning 44818 port of enip using nmap
using below coomand.

`nmap -p 44818 -A -v -Pn  [MY_IP]`

I am getting below exception.

`
Failed with exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/lmntrixdecoy/protocols/enip/enip_server.py", line 163, in handle_tcp
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 645, in run
    source=source, machine=machine, path=path, data=data, ending=ending ):
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 1226, in delegate
    for which,target in submach: # 75% of runtime is spent inside 'run'
  File "/usr/local/lib/python3.6/dist-packages/cpppo/automata.py", line 610, in run
    "%s detected no progress before finding acceptable symbol" % ( self )
AssertionError: ((byte)) detected no progress before finding acceptable symbol
`

Please help me.

Thanks.

JesseBowling, 2020-03-26:
I'd also like to know if this is something that should be fixed. In an nearly identical scenario I'm seeing these errors. I'm actually trying to figure out why connections to this port don't result in messages being send to the hpfeeds logging mechanism; no messages are sent for this port almost all of the logging for these connections is filled with tracebacks.

prashantvidja, 2020-03-27:
Hi,
I have fixed the issue by making some changes.
Please check below enip_server.py file.
[enip_server.zip](https://github.com/mushorg/conpot/files/4391831/enip_server.zip)

adepasquale, 2020-04-05:
@prashantvidja so you basically added a `session`?

Please make a pull request instead of uploading a zip file

closed 2020-04-05 by prashantvidja