mushorg/conpot #441: Gurdian

prashantvidja, 2019-03-20:


prashantvidja, 2019-03-22:
Hello,

One thing I have found that.
When i am start scanning then One connection established after that it is not disconnected even if i stop scanning. (kill nmap scanner)
That's why I think it is not possible to connect again.
I am pasting log here please check it and help me out from this issue.

> 2019-03-22 10:18:02,121 Conpot GuardianAST initialized 
> 2019-03-22 10:18:02,123 Found and enabled guardian_ast protocol.
> 2019-03-22 10:18:02,125 No proxy template found. Service will remain unconfigured/stopped.
> 2019-03-22 10:18:02,126 GuardianAST server started on: ('0.0.0.0', 10001)
> 2019-03-22 10:18:22,011 New guardian_ast session from 192.168.56.1 (44978cd2-be78-49ec-8fa8-c7052d7fd1bd)
> 2019-03-22 10:18:22,012 New GuardianAST connection from 192.168.56.1:56856. (44978cd2-be78-49ec-8fa8-c7052d7fd1bd)
> 2019-03-22 10:18:28,017 Non ^A command attempt 192.168.56.1:56856. (44978cd2-be78-49ec-8fa8-c7052d7fd1bd)
> 2019-03-22 10:18:28,017 GuardianAST client disconnected 192.168.56.1:56856. (44978cd2-be78-49ec-8fa8-c7052d7fd1bd)
> 2019-03-22 10:18:28,019 New GuardianAST connection from 192.168.56.1:56860. (44978cd2-be78-49ec-8fa8-c7052d7fd1bd)

Thanks & Regards

glaslos, 2026-09-14:
Fixed in 180c90596ca57030116802d98cd52de4ace7ecfb

closed 2026-09-14 by glaslos