mushorg/conpot #443: IPMI not generating alert in json file.

prashantvidja, 2019-03-23:
Hi,

All Protocol generate alert in json file but ipmi protocol not generating alert in json file but it is generating alert in conpot log.
For that i have check code and I found that ipmi is not using conpot/core/SessionManager, It is using its own FakesessionManager.
anybody can help me to generate log in json file?
 
Thanks.

xandfury, 2019-07-08:
Could you please share what exactly did you do? For us to replicate the issue :slightly_smiling_face:

prashantvidja, 2019-09-04:
Hello,

After set JSON log True in testing.cfg I have started conpot with the default template.
But when I am trying to scan ipmi with below command.

`nmap -sU --script ipmi-version -p 623 MY_HOST_IP`

But I am not getting JSON log in the file but for other services I get log.
I am getting below output in conpot console output.

```
2019-09-04 08:33:36,669 New IPMI traffic from ('192.168.56.1', 61611)
2019-09-04 08:33:36,671 New IPMI session initialized for client (('192.168.56.1', 61611))
2019-09-04 08:33:36,673 Connection established with ('192.168.56.1', 61611)
2019-09-04 08:33:36,675 IPMI response sent to ('192.168.56.1', 61611)
2019-09-04 08:33:36,769 Incoming IPMI traffic from ('192.168.56.1', 61612)
2019-09-04 08:33:36,905 Incoming IPMI traffic from ('192.168.56.1', 42313)
2019-09-04 08:33:36,906 Incoming IPMI traffic from ('192.168.56.1', 42313)
2019-09-04 08:33:36,908 Incoming IPMI traffic from ('192.168.56.1', 42313)

```

Please check once the enable JSON log and check.

Thanks.

glaslos, 2026-09-14:
This was fixed in #617 (`31b1f7d`), which wires IPMI attack events through `conpot.core.get_session` / `SessionManager` so they reach the JSON (and other) loggers.

`FakeSession` remains for pyghmi RMCP/IPMI protocol handling; Conpot attack-session logging is separate via `_add_event()`.

Verified locally: a Get Channel Authentication Capabilities probe (same path as `nmap --script ipmi-version`) produces `NEW_CONNECTION` and `GET_CHANNEL_AUTH_CAPABILITIES` events on the session log queue, and those events serialize with `JsonLogger`.

closed 2026-09-14 by glaslos