mushorg/conpot #444: s7comm request It's not there.

zhangxiaoyus, 2019-03-26:
除了http以外的其他协议request都是空的，请问这是我配置的问题还是本身就这样

creolis, 2019-03-26:
You have to enable the respective protocols in your template first.
Please stick to a language (and character set) that is understood by the majority of GitHub users in order to make these tickets more useful for others, otherwise I'm going to answer in my native language next time :-P

zhangxiaoyus, 2019-03-28:
Sorry, thank you for your reminder.
I started the default template, but all the request fields except HTTP are null. How to configure this? I don't understand how to use this for the first time.
[image]

glaslos, 2026-09-14:
Will be addressed by https://github.com/mushorg/conpot/pull/628

closed 2026-09-14 by glaslos