mushorg/conpot #453: unable to move the logs to syslog

dsshobha, 2019-05-02:
[syslog]
enabled = True
device = /dev/log
host = localhost
port = 514
facility = local0
socket = udp       ; udp (sends to host:port), dev (sends to device)

this is the syslog configuration,but i'm not getting the logs in syslog

glaslos, 2020-11-23:
Did you confirm you can receive a syslog on port 514? Any error messages?

ssabetan, 2021-02-23:
Below is the affected line that stops syslog from functioning correctly. In your cfg file, replace:

`socket = udp        ; udp (sends to host:port), dev (sends to device))`

with 

`socket = udp`

In syslog.py, the comparison is `str(logsocket).lower() == 'udp':`. The entire line from the configuration (including comments) are being passed to syslog.py.

closed 2026-09-14 by glaslos