mushorg/conpot #454: Improve documentation on Conpot Honeypot

mannbavara, 2019-05-15:
Which type of attack can be done on conpot honeypot? How are logs generated for specific attack?
What is structure of

creolis, 2019-05-15:
I know the state of documentation here on conpot is bad,
but is it that bad that you could not finish the sentence? ;-)

mannbavara, 2019-05-20:
@creolis What is structure of logs generated? How to export the logs to .json file?

Vingaard, 2019-05-21:
You do need to change the configuration file (-c switch) e.g conpon.cfg to; 
[json]
enabled = True

closed 2020-10-05 by mannbavara

18520165, 2021-04-15:
Hello, i have some problem about the JSON log. 

> You do need to change the configuration file (-c switch) e.g conpot.cfg to;
> [json]
> enabled = True
> 
> That would create a json log file -aka no need to export.. please see conpot/testing.cfg for inspiration

Hello, i have some problem about json log. when i change:
[json]
enable = true
and i start conpot again, it was a PermissionError like this:
[image]
So how can i solve this problem, thank you.