mushorg/conpot #455: Running conpot 0.6.0 natively on Raspberry Pi - PA

M-VdM, 2019-05-18:
Hi,

With a lot of _trial-and-error_, I managed to install the latest **conpot 0.6.0** on a **Raspberry Pi 3** _(Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux /Python 3.5.3 / pip3 19.1.1 / ...)_
Manually installing those python packages from requirements.txt - a few at the time - eventually did the trick.
Particularly cffi & cryptography proved tough to get installed ...

Now, when I try to run conpot, I get this:
BTW: [DOCs: Example Usage](url) was _not_ usefull  ;-)

**-1-** pi@raspberrypi:~ $ **sudo conpot**
WARNING:scapy.loading:Cannot read wireshark manuf database
```
                       _
   ___ ___ ___ ___ ___| |_
  |  _| . |   | . | . |  _|
  |___|___|_|_|  _|___|_|
              |_|
  Version 0.6.0
  MushMush Foundation
```
**CRITICAL:root:Can't start conpot with root. Please ref user docs for more info.**

**-2-** pi@raspberrypi:~ $ **conpot**
WARNING:scapy.loading:Cannot read wireshark manuf database
...
**Invalid arguments supplied. Please check that you pass both template and config arguments before running Conpot**

**-3-** pi@raspberrypi:~ $ **conpot -t default -c testing.cfg**
WARNING:scapy.loading:Cannot read wireshark manuf database
...
2019-05-18 14:14:21,494 Config file found!
2019-05-18 14:14:21,504 Starting Conpot using template: _/usr/local/lib/python3.5/dist-packages/conpot-0.6.0-py3.5.egg/conpot/templates/default_
2019-05-18 14:14:21,505 Starting Conpot using configuration found in: _/usr/local/lib/python3.5/dist-packages/conpot-0.6.0-py3.5.egg/conpot/testing.cfg_
2019-05-18 14:14:21,554 Can't find the temp directory. _Conpot VFS would be kept at : /usr/local/lib/python3.5/dist-packages/conpot-0.6.0-py3.5.egg/conpot/ConpotTempFS
ERROR:root:Can't start conpot with default file system
NoneType: None
2019-05-18 14:14:21,555 Can't start conpot with default file system
NoneType: None_

-4- pi@raspberrypi:~ $ **conpot -t default -c testing.cfg --temp_dir ~/conpot**
WARNING:scapy.loading:Cannot read wireshark manuf database
...
2019-05-18 14:20:03,447 Config file found!
2019-05-18 14:20:03,457 Starting Conpot using template: /usr/local/lib/python3.5/dist-packages/conpot-0.6.0-py3.5.egg/conpot/templates/default
2019-05-18 14:20:03,458 Starting Conpot using configuration found in: /usr/local/lib/python3.5/dist-packages/conpot-0.6.0-py3.5.egg/conpot/testing.cfg
_ERROR:root:Can't start conpot with default file system
NoneType: None
2019-05-18 14:20:03,506 Can't start conpot with default file system
NoneType: None_

==> What arguments do I need to pass to get conpot running ?
Thanks in advance for amending the documentation.

Kind Regards,
M_VdM.

xandfury, 2019-05-20:
Sorry about the confusion. 
`conpot -f --template default` should work :slightly_smiling_face:

xandfury, 2019-07-08:
I hope you issue has been resolved :slightly_smiling_face:

closed 2019-07-08 by xandfury