mushorg/conpot #456: Stream log file from CONPOT to a remote server?

erickonavarro, 2019-05-26:
I want to be able to stream the log events when running CONPOT to a remote server. What is the easiest way to do this?

I'm currently working on a project that involves a Honeypot which is generating events on a .log file. I want to send these events to a remote server. What is the easiest way to do this?

I've been doing some research and tried to do it with rsyslog using imfile but It's not working for me; I'm not sure if I'm doing it the right way though.

This is what I added to my rsyslog.conf file on my client

```
    *.* @@<ipaddress>:514
    /etc/rsyslog.conf$
    $ModLoad imfile
    $InputFileName /var/log/conpot/conpot.log
    $InputFileTag CONPOT
    $InputFileSaverity Info
    $InputFileFacility local3
    $InputRunFileMonitor
    local3.* @@<ipaddress>:514
```
This is what I added to my rsyslog.conf file on my server

```
    module(load="imudp")
    input(type="imudp" port="514")
    module(load="imklog" permitnonkernelfacility="on")
    $template RemoteLogs, "/var/log/$HOSTNAMES%/%PROGRAMNAME%.log"
    *.* ?RemoteLogs
    & ~
```
That's all I have added to my conf files, what I'm I missing?

I expect to get all the events from ~/conpot.log to my rsyslog server but I do not seem to get them

xandfury, 2019-05-30:
Glad to hear things worked out for you :slightly_smiling_face:

closed 2019-05-30 by xandfury

creolis, 2019-05-30:
Please note that

`InputFileSaverity` will not work :) -> `InputFileSeverity`