mushorg/conpot #457: docker-compose v1.22.0 conpot build fails

samrreynolds4, 2019-06-10:
Hello,

I was trying to build conpot using docker-compose and it looks like the tests are failing:
```
  /home/conpot/.local/lib/python3.6/site-packages/cpppo/automata.py:645: DeprecationWarning: generator 'state.delegate' raised StopIteration
    source=source, machine=machine, path=path, data=data, ending=ending ):

conpot/tests/test_enip_server.py::TestENIPServer::test_read_tags
conpot/tests/test_enip_server.py::TestENIPServer::test_write_tags
  /home/conpot/.local/lib/python3.6/site-packages/cpppo/server/enip/client.py:1190: DeprecationWarning: generator 'connector.collect' raised StopIteration
    issued, self.collect( timeout=timeout )): # must be "lazy" zip!

conpot/tests/test_kamstrup_meter_protocol.py::TestKamstrup::test_request_get_register
  /opt/conpot/conpot/core/databus.py:70: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
    assert len(inspect.getargspec(callback)[0])  # depreciated in py3.5, un-depreciated in py3.6

conpot/tests/test_tftp.py::TestTFTPServer::test_mkdir_upload
conpot/tests/test_tftp.py::TestTFTPServer::test_tftp_upload
conpot/tests/test_vfs.py::TestFileSystem::test_copyfile
conpot/tests/test_vfs.py::TestFileSystem::test_movefile
conpot/tests/test_vfs.py::TestFileSystem::test_open_file
conpot/tests/test_vfs.py::TestFileSystem::test_openbin_file
conpot/tests/test_vfs.py::TestSubFileSystem::test_open_file
  /home/conpot/.local/lib/python3.6/site-packages/fs/base.py:76: DeprecationWarning: method 'gettext' has been deprecated, please rename to 'readtext'
    DeprecationWarning,

-- Docs: https://docs.pytest.org/en/latest/warnings.html
======= 2 failed, 160 passed, 5 skipped, 1425 warnings in 161.44 seconds =======
ERROR: Service 'conpot' failed to build: The command '/bin/sh -c py.test -v' returned a non-zero code: 1

```

Docker version: 18.09.6, build 481bc77156
docker-compose version: 1.22.0, build f46880fe

palif, 2020-01-18:
2 of these test have failed in the building of the docker container and that is why the docker was not able to run the container.

Sometimes the test could be outdated/old and needs to be upgraded due. Therefore, a solution could be to comment out the line that runs the test in the Dockerfile (probably in line 24; "RUN py.test -v").

Hopefully that will work.

glaslos, 2020-11-09:
Should be fixed with #522

closed 2020-11-09 by glaslos