mushorg/conpot #458: Unable to build from source

jeffdougherty, 2019-07-05:
Summary: I am unable to build and run conpot from source on my Ubuntu 18.04.01 machine.  (See bottom for detailed version info).  I can run conpot by pulling from honeynet/conpot, but trying to build locally either with the `docker` or `docker-compose` methods from the README fails due to failing two of the built-in tests.  It's always the same two:

conpot/tests/test_enip_server.py::TestENIPServer::test_read_tags FAILED  [  8%]
conpot/tests/test_enip_server.py::TestENIPServer::test_write_tags FAILED [  9%]

I've tried both installation methods, and also tried installing on Debian 9 with similar results so I'm pretty confident it's not an Ubuntu issue.  I'd like to do some development work with conpot, so building from the honeynet image is not really adequate for my needs.  I'm also reasonably confident this isn't a permissions issue since I was using `sudo` to run the build commands.

I'm attaching a text file of my stdout/stderr output during an attempt to build using `docker` on Ubuntu.  There's a lot of detailed trace that doesn't mean much to me, perhaps it will mean something to somebody here.

Help?
[docker-buildlog.txt](https://github.com/mushorg/conpot/files/3363773/docker-buildlog.txt)

OS: Ubuntu 18.04.01
Docker version: 18.09.07
Docker-compose version output: 
docker-compose version 1.17.1, build unknown
docker-py version: 2.5.1
CPython version: 2.7.15+
OpenSSL version: OpenSSL 1.1.1  11 Sep 2018

jnzhuang, 2019-07-08:
I also encountered these two failures, and what I did was just to comment out the line "RUN py.test -v", and it can be installed and run. As long as you do not need ENIP, you can just circumvent the issue. But  I also yearn professional solutions to these two failures.

jeffdougherty, 2019-07-08:
@jnzhuang Have you tried doing your own local development work, and if so have your changes been included when you build locally?  I notice that during the `docker build` step it pulls an image from Github, so I'm wondering if my changes would even be included.  Any insight?

jnzhuang, 2019-07-08:
> @jnzhuang Have you tried doing your own local development work, and if so have your changes been included when you build locally? I notice that during the `docker build` step it pulls an image from Github, so I'm wondering if my changes would even be included. Any insight?

Actually I do not quite catch your question. But if you want to know if your local changes to xml file take effect in the docker, my experience tells me no. To solve this, I have installed conpot on a virtual machine (ubuntu OS) without docker.

jeffdougherty, 2019-07-08:
@jnzhuang Can you share what your steps were for that?  It's what I've been trying to do and I've been completely unable to get anywhere.

jnzhuang, 2019-07-08:
Sure. 
(1) "git clone" the code to your virtual ubuntu-OS machine (not necessarily in /opt folder)
(2) install python3 if you have not (do use python3, never use python 2)
(3) refer to https://conpot.readthedocs.io/en/latest/installation/ubuntu.html#installation, edit source.list, install dependencies.
(4) install python dependencies required. They can be found in the requirement.txt file in conpot folder. (reminder: to install, do use pip3 instead of pip)
(5) from the conpot folder, run "sudo python3 setup.py install" (documentation says python, no, use python3. Also we need sudo, because we are not in docker)
(6) use "which conpot" to see if the executable file is properly generated.
(7) run "conpot -f --tempate default", but there are probably some issues remaining.
(8) fix the issues as hinted by the error info. (mostly some permission denied problems. To fix, use something like "sudo chown", "sudo chgrp", and/or "sudo chmod +777", do not use "sudo conpot" to run conpot as it is not recommended).

As you remove error info, you can run it properly. Basic settings can be modified in xml files. But you have to "sudo python3 setup.py install" again to make them take effect, and also you have to reset the previlage for permissions. You may consider write them in a sinlge bash script file.

jeffdougherty, 2019-07-08:
@jnzhuang You are my new hero.  Finally got it to build and run!

For anyone else who sees this issue: I ran into permissions problems with conpot's temporary FS, and with creating a socket for SNMP (161/UDP).  For the first, I solved it by running with `--temp_dir /tmp` so it's not trying to make its temporary filesystem in the Python egg.  For the second, I used `authbind`, a utility that works like `sudo` but only gives authorization to bind ports instead of changing uid to root.  As @jnzhaung mentioned above, bad things happen if you try to run conpot as root.

jnzhuang, 2019-07-08:
@jeffdougherty Glad that it works for you too. Yes, I also use authbind. Good point.

xandfury, 2019-07-08:
> "sudo chmod +777", do not use "sudo conpot

@jeffdougherty @jnzhuang I do not think something like `chmod +777` is a good idea when running a honeypot.
Please try fresh install using this guide: https://conpot.readthedocs.io/en/latest/installation/virtualenv.html. Conpot can be installed via pip. No `sudo`/`chmod`/`chgrp`/`chown` required. :slightly_smiling_face:

creolis, 2019-07-16:
[image]

jeffdougherty, 2019-07-17:
@xandfury @creolis Your points are well taken.  A couple notes:

1. Most of the permission problems were solved by using --temp_dir=/tmp, so conpot wasn't trying to make its temporary FS in a place where that kind of activity wasn't intended.

2. As I recall, I did have to loosen some file permissions during install.  However, I also kept notes of what I changed and what the original owner and permissions were, and made sure to revert before running.

glaslos, 2020-11-09:
This will be fixed with #522

closed 2020-11-09 by glaslos